arduino/arduino-ide #1773: "Preferences" dialog scrollability not easily discoverable by user

per1234, 2022-12-15:
### Describe the problem

The dimensions of the "**Preferences**" dialog is limited by the Arduino IDE window dimensions.

If the window dimensions are small, the contents of the dialog may not fit within the available space. In this case, the dialog is scrollable, which allows the user to access the settings that fall outside the visible area of the dialog.

🐛 The ability to scroll the dialog is difficult for the user to discover. These users are given the impression that some preferences are missing from the dialog due to a bug in Arduino IDE.

### To reproduce

1. Resize the Arduino IDE window to the minimum vertical size.
1. Select **File > Preferences** from the Arduino IDE menus.
   The preferences farthest down in the dialog are not visible.
1. Move the mouse pointer to an area of the screen not over the dialog:
   [image]
   🐛 There is no indication at all that the dialog is scrollable:
1. Hover the mouse pointer over the dialog, but not over the scrollbar area:
   [image]
   🐛 A scrollbar appears, but the non-standard design causes the nature of the UI element to be non-obvious to the average user:
      - Rounded "thumb"
      - Thumb is significantly less than standard width
      - No "trough"
      - Scrollbar overlaps the horizontal rule of the tab bar
      - There is no spacing between the right side of the "**BROWSE**" button and the scrollbar
1. Hover the mouse pointer over the the scrollbar area of the dialog:
   [image]
   🙂 The nature of the UI element is now fairly obvious to the average user:
   - There is a trough
   - The scrollbar has a standard width

### Expected behavior

It is obvious to the user when a dialog is scrollable.

### Arduino IDE version

7c86f1f

### Operating system

Windows

### Operating system version

10

### Additional context

Originally reported at https://forum.arduino.cc/t/missing-additional-board-manager-urls-field-in-preferences/1055935/1

Additional reports:

- https://forum.arduino.cc/t/arduino-core-available-for-the-ai-thinker-bl602/1064372/5
- https://forum.arduino.cc/t/missing-additional-board-manager-urls-field-in-preferences/1055935/7

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

seaxwi, 2023-04-19:
For me (on macOS), resizing the Arduino IDE window to it's minimum size does not replicate the issue:

[image]

However, the same problem occurs if I increase the **Interface scale** to 110%.

[image]

This is the scrollable area with Interface scale set to 120% percent:

[image]

I've also received a small amount of Help Center article feedback with complaints about the "Additional boards manager URLs" field missing, that may be caused by this issue.

PaulStoffregen, 2024-08-18:
This problem came up [recently on the Teensy forum](https://forum.pjrc.com/index.php?threads/cant-install-teensyduino-into-arduino-ide-2-3-2.75658/#post-348085), with a new user (first ever use of Arduino IDE) unable to find the Additional boards manager URLs.  They were using Arduino IDE 2.3.2.

[image]

I know this is an older issue, but really hoping it can get attention for a fix soon.  It really does trip up first time users.