arduino/arduino-ide #1775: "Preferences" dialog shows proxy is configured as SOCKS after user configures HTTP proxy

per1234, 2022-12-16:
### Describe the problem

Users may access the Internet via a [proxy server](https://en.wikipedia.org/wiki/Proxy_server). Arduino IDE can be configured for this use case via the "**Network**" tab of the "**Preferences**" dialog.

Two types of proxies are supported:

- HTTP
- [SOCKS](https://en.wikipedia.org/wiki/Proxy_server#SOCKS_proxy)

The proxy type is selected via a radio button on the dialog.

🐛 The radio button is set to "**SOCKS**" after the IDE is reloaded even though the user selected the "**HTTP**" radio button in the "**Preferences**" dialog when they configured the proxy preferences.

### To reproduce

1. Select **File > Preferences...** from the Arduino IDE menus.
1. Click the "**Network**" tab.
1. Select the "**Manual proxy configuration**" radio button.
1. Configure proxy as follows:
   - Select the "**HTTP**" radio button.
   - Set "**Host name**" to `example.com`
   - Set "**Port number**" to `42`
   - Set "**Username**" to `foo`
   - Set "**Password**" to `bar`
1. Click the <kbd>**OK**</kbd> button.
   The index download process may now fail. This is expected due to the use of an dummy proxy configuration for the demo.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start Arduino IDE.
1. Select **File > Preferences...** from the Arduino IDE menus.
1. Click the "**Network**" tab.

🐛 The radio button is set to "**SOCKS**":

[image]

### Expected behavior

Proxy type radio button in the "**Preferences**" dialog always correctly reflects the proxy type that was configured.

### Arduino IDE version

908ec4c

### Operating system

Windows

### Operating system version

10

### Additional context

The "**Preferences**" dialog shows the correct "**HTTP**" radio button if the dialog is opened again after configuring the proxy but before restarting Arduino IDE. The fault only occurs after a restart.

---

The `network.proxy` key in `~/.arduinoIDE/arduino-cli.yaml` does have the `http` URL scheme as expected even though the "**Preferences**" dialog indicates it is a SOCKS proxy (for which Arduino IDE [currently uses the `socks` URL scheme](https://github.com/arduino/arduino-ide/issues/1769)):

```yaml
network:
  proxy: http://foo:bar@example.com:42/
```

---

Originally reported by @wh201906 at https://github.com/arduino/arduino-ide/issues/1769

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

wh201906, 2022-12-17:
This is what I found. 
I put a breakpoint in renderProxySettings() and I found the `this.state.network.protocol` is `https:` rather than `http`
(I printed `this.state.network` in the console. You can find the `protocol` in the bottom of the image.)
[image]
Since the SOCKS radiobutton loads after the HTTP radiobutton, it find `https:` doesn't match `http`, then set `checked=true`
I guess this is the reason?

wh201906, 2022-12-17:
If I click HTTP button in Preferences->Network, `this.state.network` is `http` rather than `http:`
[image]

wh201906, 2022-12-17:
I think PR #1778 will fix this.

PR #1778 (2022-12-21): Remove trailing colon when parsing the protocol from URL
### Motivation
This PR should fix #1775

### Change description
Remove the trailing colon when parsing the protocol in URL string

### Other information

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / updated (for bug fixes / features)
closed 2022-12-21 by kittaakos