arduino/arduino-ide #1779: Collect all temporary files in a temporary directory

sterretjeToo, 2022-12-18:
### Describe the request

IDE 2.0 leaves a lot of directories files behind in (Windows system) `C:\Users\yourUsername\AppData\Local\Temp`. Request to collect them all in one dedicated Arduino directory.

If the cause is a virus scanner that prevents cleanup, it's safer to just exclude on directory instead of the complete directory.
If the cause is a sloppy IDE, it's easier to manually clean up at occasion.

### Describe the current behavior

Directories and files that can be found in `C:\Users\yourUsername\AppData\Local\Temp`

* Files `C:\Users\yourUsername\AppData\Local\Temp\.arduinoIDE-unsaved*`.
* Directories `C:\Users\yourUsername\AppData\Local\Temp\longNumber` (e.g. 5577006791947779410206689246) with files `index.gz`, `index.json` and `index.json.sig`.
* Directories `C:\Users\yourUsername\AppData\Local\Temp\longNumber\arduino-ide2-longHexNumber`; directories are empty
* Directories `C:\Users\yourUsername\AppData\Local\Temp\arduino-language-serverNumber (e.g. `arduino-language-server525288423`).
* Directories `C:\Users\yourUsername\AppData\Local\Temp\arduino-sketch-longHexNumber` (e.g. `arduino-sketch-07DD36AA0DA5BBED7460DCF645FC5952`).
* Files `C:\Users\yourUsername\AppData\Local\Temp\dd_updateconfiguration_yyyyMMddHHmmss.log`
* Files `C:\Users\yourUsername\AppData\Local\Temp\gdb-server-console-number.log` (e.g. `gdb-server-console-5992.log`).
* Files `C:\Users\yourUsername\AppData\Local\Temp\system-includes-number.clangd (e.g. `system-includes-75256d.clangd`)
* ico files

Note
Above list from IDE 2.0.3; I haven't consistently used the nightly build to be sure if the list has not changed significantly but a number of the above directories and files are certainly there.

### Arduino IDE version

arduino-ide_nightly-20221217_Windows_64bit

### Operating system

Windows

### Operating system version

Win10 Home

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

sterretjeToo, 2022-12-20:
@kittaakos and @per1234 

As I don't know what is all involved, I'll leave it up to you guys to add similar feature requests for all the other applications in the tool chain ;)

wh201906, 2022-12-22:
I guess the `.arduinoIDE-unsaved*` is created by Arduino IDE?
https://github.com/arduino/arduino-ide/blob/76f9f635d8558f399423f64c0c709e3a53bdcd03/arduino-ide-extension/src/node/sketches-service-impl.ts#L515-L537

per1234, 2023-01-02:
In addition to the two classes of folders mentioned above, three others are created by the Arduino IDE application:

- `github-remote`
  - The folder is empty
  - Origin: https://github.com/eclipse-theia/theia/blob/v1.31.1/packages/plugin-ext/src/main/node/plugin-github-resolver.ts#L39
- Name: `http-remote`
  - The folder is empty
  - Origin: https://github.com/eclipse-theia/theia/blob/v1.31.1/packages/plugin-ext/src/main/node/plugin-http-resolver.ts#L37
- Name: `theia_upload`
  - The folder is empty
  - Origin: https://github.com/eclipse-theia/theia/blob/v1.31.1/packages/filesystem/src/node/node-file-upload-service.ts#L55