arduino/arduino-ide #1783: Access is denied for com port  on Windows 11 with Nano

maciejmiklas, 2022-12-23:
I've been using Arduino IDE 2.0.3 with Nano without issues when uploading on Ubuntu. Now I've tried it on Windows 11, and most of the time, it gives me this error:
``` 
avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\macie\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "\\.\COM5": Access is denied.
```

But I can see logs from the serial monitor in Arduino IDE -> serial monitor. Also, I have no problems uploading sketches on Arduino IDE 1.8.19 from Microsoft Store on the same system. I can quit 2.0.3, start 1.8.19, and it's working immediately.

I do not have an official board because I've soldiered Atmega onto my build. I've tried official Mega and Uno and have had no issues, so that this problem might be related to a particular diver or USB to serial chip.

Wezz19, 2023-01-09:
Not sure if it's actuality related to your problem, but I got this after I had changed the COM port in device manager. After rebooting the computer I could use the COM port normally again.

Is this still happening for you?

maciejmiklas, 2023-01-10:
Yes, I can work for several hours on 1.8 without a single issue, and this problem occurs immediately after switching to 2.0

per1234, 2023-01-26:
Hi @maciejmiklas. Thanks for your report.

Does the error still occur if you close the **Serial Monitor** tab in the bottom panel of the Arduino IDE window before uploading?

You can do that by clicking the **X** icon on the tab:

[image]

Arduino IDE automatically closes the port in Serial Monitor and Serial Plotter when uploading, so this should not be necessary. I'm only trying to determine the exact conditions the fault occurs under.

maciejmiklas, 2023-01-27:
I did that and have had no problems so far.

per1234, 2023-01-27:
Thanks for checking @maciejmiklas.

I am able to reproduce this fault, but only under specific conditions (FT232 chip, Windows, using Serial Monitor/Plotter) and I wanted to make sure the fault you are experiencing also depends on those conditions.

Knowing the required conditions and being able to reproduce it should make the preparation and validation of a fix for the bug much easier.

maciejmiklas, 2023-01-27:
I do not have this problem with the original UNO, only when using a custom board with USB->Serial adapter:

[image]

ananzjo, 2023-05-03:
Still Have the Same Problem Although Serial Monitor & Serial Plotter are both closed , this is very frustrating 

any advice please 

`FQBN: arduino:avr:nano
Using board 'nano' from platform in folder: C:\Users\ananz.CAD\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\ananz.CAD\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6

Detecting libraries used...
"C:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\eightanaloginputs" "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D\\sketch\\frank-Zaid.ino.cpp" -o nul
Generating function prototypes...
"C:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\eightanaloginputs" "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D\\sketch\\frank-Zaid.ino.cpp" -o "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\eightanaloginputs" "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D\\sketch\\frank-Zaid.ino.cpp" -o "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D\\sketch\\frank-Zaid.ino.cpp.o"
Compiling libraries...
Compiling core...
Using precompiled core: C:\Users\ananz.CAD\AppData\Local\Temp\arduino\cores\arduino_avr_nano_cpu_atmega328_ff382133097821e9b3511dee9c511050\core.a
Linking everything together...
"C:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-gcc" -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D/frank-Zaid.ino.elf" "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D\\sketch\\frank-Zaid.ino.cpp.o" "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D/..\\..\\cores\\arduino_avr_nano_cpu_atmega328_ff382133097821e9b3511dee9c511050\\core.a" "-LC:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D" -lm
"C:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D/frank-Zaid.ino.elf" "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D/frank-Zaid.ino.eep"
"C:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-objcopy" -O ihex -R .eeprom "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D/frank-Zaid.ino.elf" "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D/frank-Zaid.ino.hex"

"C:\\Users\\ananz.CAD\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-size" -A "C:\\Users\\ananz.CAD\\AppData\\Local\\Temp\\arduino\\sketches\\51312D1A002CB6593C184EABDDDB0A2D/frank-Zaid.ino.elf"
Sketch uses 3418 bytes (11%) of program storage space. Maximum is 30720 bytes.
[... cut ...]
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\ananz.CAD\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM5
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: ser_open(): can't open device "\\.\COM5": Access is denied.

avrdude done.  Thank you.

Failed uploading: uploading error: exit status 1`

per1234, 2023-05-03:
Hi @ananzjo

> any advice please

This is not an appropriate place to request advice. This issue is only used by the developers to track the bug. If you would like assistance with fixing the upload error (which has multiple potential causes other than the one tracked here), please post on Arduino Forum:

https://forum.arduino.cc/

I'm sure we'll be able to help you out over there.

jenschr, 2023-06-30:
I just spent an evening helping a non-technical person (cosplayer) finding a solution to this problem. Closing the Serial Monitor or installing an older version of the IDE does the trick. This person was on Win 11 using Arduino 2.1.0 IDE with an original Arduino Nano. This is a very common setup, so this will affect many users.

It puzzles me a bit that this bug still has not been assigned to a milestone given that it was reported 6+ months ago. Are there problems reproducing it? Anything I can do to help resolve it?

kittaakos, 2023-08-18:
I investigated, and the problem cannot be fixed without CLI changes/support.

IDE2 creates a gRPC [bidirectional stream](https://grpc.io/docs/what-is-grpc/core-concepts/#bidirectional-streaming-rpc) via the [`Monitor`](https://github.com/arduino/arduino-cli/blob/38479dc706f27b2640cabad0bf39cd24e4f049c7/rpc/cc/arduino/cli/commands/v1/commands.proto#L177) method.

BeforeIDE2 does an upload, all related monitor connections are closed. IDE2 closes the monitor connection by ending the bidirectional stream and waiting for the OK signal:

https://github.com/arduino/arduino-ide/blob/69ae38effaf950ae6fba83617b08b1cc0bc0a40d/arduino-ide-extension/src/node/monitor-service.ts#L435

I see a race condition; if I put a breakpoint into the code after closing the bidirectional stream (and let the OS clean up the resources), I do not have the `access is denied` error.

```
2023-08-18T14:19:15.694Z daemon INFO time="2023-08-18T16:19:15+02:00" level=info msg="sending command" command=CLOSE monitor=serial-monitor
2023-08-18T14:19:15.705Z daemon INFO time="2023-08-18T16:19:15+02:00" level=info msg="received message" error=false event_type=close message=OK monitor=serial-monitor
2023-08-18T14:19:15.707Z daemon INFO time="2023-08-18T16:19:15+02:00" level=info msg="sending command" command=QUIT monitor=serial-monitor
time="2023-08-18T16:19:15+02:00" level=info msg="received message" error=false event_type=quit message=OK monitor=serial-monitor
2023-08-18T14:19:15.710Z daemon INFO time="2023-08-18T16:19:15+02:00" level=info msg="Killing monitor process" monitor=serial-monitor
2023-08-18T14:19:15.711Z daemon INFO time="2023-08-18T16:19:15+02:00" level=error msg="Sent kill signal" error="TerminateProcess: Access is denied." monitor=serial-monitor
2023-08-18T14:19:15.719Z daemon INFO time="2023-08-18T16:19:15+02:00" level=error msg="stopped decode loop: EOF" monitor=serial-monitor
2023-08-18T14:19:15.736Z daemon INFO time="2023-08-18T16:19:15+02:00" level=info msg="Monitor process killed" monitor=serial-monitor
```

https://github.com/arduino/arduino-ide/assets/1405703/ef510025-ffd7-4368-af39-64c93e2d9305

CLI should provide a way to terminate the monitor connection gracefully. Calling this API and receiving an OK ack means the port is free.

This is what is happening:
 - client (IDE2) ends the stream
 - server (CLI) receives the end of the stream, kills the monitor process
 - the stream is already closed on the client, it does not wait for the server, the server is still killing the processes (maybe),
 - the client starts an upload
 - 💥 

This is what should happen:
 - client (IDE2) send a `STOP` request via the open bidirectional stream,
 - server receives the `STOP` request (the server is allowed to ignore any other requests received after the `STOP`) and closes the monitor,
 - server sends a `STOP ACK` equivalent,
 - client ends the connection,
 - server receives the end of stream (noop, the monitor process is already killed)

This is just one of the possible solutions, but the main idea is that IDE2 or any other CLI client might need this way of terminating the monitor.

per1234, 2023-08-19:
Thanks for taking the time to make an investigation @kittaakos! I am sorry it wasn't so simple as I had hoped.

Since you are now most familiar with the problem and the work that will be required to resolve it, would you mind reporting this to the Arduino CLI developers by submitting an issue to their repo when you get the time?

If work is needed in both the Arduino IDE and Arduino CLI codebases, then I would suggest we keep this one here to track the Arduino IDE work and create a separate one in the Arduino CLI repo to track the work needed on that codebase.

ubidefeo, 2023-08-19:
@kittaakos , I agree with @per1234 

Please, report to the CLI team and let's do things right even if it shifts development a little.
This is something that should be addressed in the next release :)

jmcsch7, 2024-06-05:
I am experiencing this error too  "avrdude: ser_open(): can't open device "\\.\COM9": Access is denied.". With the Arduino IDE 2, and Dfrobot Arduino Leonardo (DFR0282).

per1234, 2024-06-05:
Hi @jenschr. This issue is specific to boards that use an FTDI FT232R USB chip. There are other unrelated causes of the "Access is denied." error. The DFRobot Leonardo does not use the FT232R chip so your error must be caused by something else.

If you haven't done so already, please make a post on Arduino Forum:

https://forum.arduino.cc/c/using-arduino/avrdude-stk500-bootloader-issues/81

We'll be happy to help you solve the problem over there.

samohd, 2024-06-28:
I have encountered this issue as well. I have two (clone?) UNO boards. Visually they look identical. One is CH340 based (Driver Provider: wch.cn). The other is FTDI (Driver Provider: FTDI) based. The CH340 does NOT have this problem. With the serial monitor OPEN, I can upload new code and the serial monitor disconnects and reconnects automatically. With the FTDI one I have to manually close/reopen Serial Monitor or I get "avrdude: ser_open(): can't open device "\\.\COM#": Access is denied" error.

While on this topic, if someone gets around to fixing this... it would be REALLY nice if when the Serial Monitor is reconnected it automatically cleared the display... particularly since there's no way to do it programatically. Or if there was a Preferences setting for this would be fine too.