arduino/arduino-ide #1787: Linux Version does not show USB ports

s0170071, 2022-12-26:
### Describe the problem

apparently the current user has to be added to the dialout group.
This has been the case for some time, however, the IDE now does not even offer a greyed-out  the tools->port setting.
I suggest to add a pop up note when the port menu item is clicked when there are no ports available. 

### To reproduce

fresh installed Linux mint and the IDE 2.03 as appimage.

### Expected behavior

there should be a check upon program startup if user is a member of the dialout group.

### Arduino IDE version

2.03

### Operating system

Linux

### Operating system version

20.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-12-27:
Closing as duplicate of https://github.com/arduino/arduino-ide/issues/655, which has already been fixed by https://github.com/arduino/arduino-ide/pull/1774

closed 2022-12-27 by per1234

s0170071, 2022-12-27:
Agreed that that grayed out menu is a duplicate.
But checking for dialout membership would geatly improve the user experience.I just love it when you download the IDE and it works out of the box.
Same goes for the Espressif boards manager url. Why isnt that one added as default?