arduino/arduino-ide #1788: CH340 driver very difficult to install under UEFI Linux

s0170071, 2022-12-26:
### Describe the problem

Since Linux Mint version 20.3 the CH340 driver[ is not included ](https://forums.linuxmint.com/viewtopic.php?t=380132)in the kernel (version >5.3). 

Normally, the driver just needs to be [downloaded, compiled and added to the system: ](https://github.com/juliagoda/CH341SER) That is fairly straight forward and acceptable. 

If the system is in UEFI mode, [the driver needs to be signed](https://ubuntu.com/blog/how-to-sign-things-for-secure-boot).

It would be nice if the IDE came with a functionality to handle this. The way it is now, the IDE is de facto unusable with USB boards that use CH340, e.g. ESP8266 Wemos D1 mini.

### To reproduce

Use a out-of-the-box Linux mint and try to get a CH340 based board programmed.

### Expected behavior

As support for CH340 has been removed from the kernel, the driver needs to be installed manually. Additionally it need to be signed when using UEFI. IMHO that is too difficult for the average user and contradicts the  works out of the box / easy to use concept of the IDE.

I would expect the IDE to handle this.

### Arduino IDE version

2.0.3

### Operating system

Linux

### Operating system version

Linux Mint 21

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-12-27:
Hi @s0170071. Thanks for your suggestion. There is nothing that can be done in the Arduino IDE codebase to fix or mitigate this problem. The problem affects the upload tool directly, and Arduino IDE only indirectly.

Because Arduino has created [a flexible boards platform framework](https://arduino.github.io/arduino-cli/latest/platform-specification/) which can be used to add support for any microcontroller and board to Arduino IDE, using any arbitrary toolchain, Arduino IDE (or more specifically its [**Arduino CLI**](https://arduino.github.io/arduino-cli/latest/) helper tool) has no knowledge of what specifically caused an upload to fail; only that the upload tool returned a non-zero exit status (and the status code also does not provide any indication).

These upload tools (e.g., [**AVRDUDE**](https://github.com/avrdudes/avrdude) are not maintained by Arduino, so Arduino can't implement some sort of special check for this condition in the tool. I think doing something like that would be very difficult even for the developers of the tools to implement because the tool only sees that access is denied to the port, which could be caused by any arbitrary application (e.g., a serial terminal program) having the port open.

The place for the fix is at the source of the problem: the BRLTTY project. Please report this irresponsible treatment of serial ports by BRLTTY to the maintainers of that project:

https://brltty.app/contact.html

---

The place where I do think Arduino can mitigate this common (the scope is actually larger than the specific combination of Linux Mint + CH340 you mention here) source of problems for users is via documentation. I think it would be a good idea to document this cause of the error and the fix in an [**Arduino Help Center**](https://support.arduino.cc/hc/en-us) article. I see there are some related articles:

- https://support.arduino.cc/hc/en-us/articles/360016495679-Fix-port-access-on-Linux
- https://support.arduino.cc/hc/en-us/articles/4403365313810-Errors-when-uploading-a-sketch
- https://support.arduino.cc/hc/en-us/articles/4401874331410

but nothing about BRLTTY specifically. For this reason, I have transferred the issue to the repository where the Help Center content is maintained.

closed 2022-12-27 by per1234

DaveMielke, 2022-12-27:
For purposes of introduction, I'm the dude who has the final word concerning the brltty project. I've received several emails drawing my attention to the fact that brltty is stealing generic USB to serial adapters, like the CH340, from those who'd like to use them for other purposes. Yes, that buck does stop at me. I've just been asked to comment on this issue here, supposedly because that'd be a way to contact most of you at once.

Some braille device manufacturers, especially recently, have opted to use generic internal USB to serial adapters without customizing them (vendor/product identifiers, manufacturer/product strings, etc). Nevertheless, brltty still needs to be able to recognize even these devices because, without an operational braille device, a braille user (who is most probably blind) effectively has no screen. It'd be just like any of you sighted users booting your computer and then finding yourself without a screen while you try to figure out why it isn't working and how to get it working.

I'm guessing that those of you who are finding this issue to still be a problem have a release of brltty earlier than 6.5 installed on your computer. In 6.5, we implemented what we hoped would be a reasonable resolution to it. We split out the rules for claiming generic USB to serial adapters out into a separate udev rules file which needs to be installed separately from the main package.

You can tell which version of brltty you have by using its -V (or --version) command line option. If it's earlier than 6.5 then I suggest upgrading it. If it's 6.5 or later then I'm guessing that that optional udev rules package is installed and should be removed.

s0170071, 2022-12-27:
@DaveMielke thanks for stopping by here. 
My syslog reads  `BRLTTY 6.4 rev BRLTTY-6.4 `. It does not provide the third digit of its version number. But I assume it is 6.4.4 because this is what I get when I re-install it. 
During installation, the console says: 
`http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 brltty amd64 6.4-4ubuntu3 [1365 kB]
`
I could test 6.5 to see if the issue is gone, but I have no idea how I would get that latest version. Apparently the ubuntu  repositories (and linux mint for that matter) still provide v6.4.4 as latest version.

DaveMielke, 2022-12-27:
[quoted lines by s0170071 on 2022/12/27 at 18:11 +0000]

>My syslog reads  `BRLTTY 6.4 rev BRLTTY-6.4 `. It does not provide the third digit of its version number. 

We don't often have releases with a third digit, so you can assume 0.

>But I assume it is 6.4.4 because this is what I get when I re-install it. 
>During installation, the console says: 
>`http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 brltty amd64 6.4-4ubuntu3 [1365 kB]
>`

that .4 would've been added by the Ubuntu (or maybe the Debian) people. They do that kind of thing when they pick up commits from after 6.4 because they think they're important enough.

>I could test 6.5 to see if the issue is gone, but I have no idea how I would get that latest version. 

Nor do I as I use Fedora.

>Apparently the ubuntu  repositories (and linux mint for that matter) still provide v6.4.4 as latest version. 

6.5 was released this past June. Maybe you need to point your sources.list (I think that's what Debian calls it) to some new, testing, or experimental repository, or something like that.

Another thing you should be able to do is to disable the brltty service. Again, since I'm not a Debian user, I don't know, off the top of my head, how to do that on a Debian-based system.

Igorpereirag, 2024-03-02:
bom trabalho

ElectricRCAircraftGuy, 2025-07-15:
> Another thing you should be able to do is to disable the brltty service. 

From [my answer here](https://unix.stackexchange.com/a/797940/114401) (see also [here](https://github.com/arduino/help-center-content/issues/155#issuecomment-3070783754) and [here](https://github.com/ElectricRCAircraftGuy/eRCaGuy_dotfiles/tree/master/arduino)):

```bash
# Stop the brltty service for this session:
sudo systemctl stop brltty
# Disable it from starting up again on boot:
sudo systemctl disable brltty
# Check the status of the brltty service to ensure it is stopped and disabled: 
sudo systemctl status brltty
```