arduino/arduino-ide #1789: Avoid recompiling when uploading multiple times without making changes

raffaeler, 2022-12-28:
### Describe the problem

When I successfully compile and upload a sketch, a first board is successfully flashed.
If I change the board and press the upload button, the IDE recompiles everything before starting the new upload.
This new recompilation is NOT needed as no changes were made to the source files.

### To reproduce

1. Create a sketch and press the upload button
2. The binary file is compiled and uploaded to a board (I am using ESP32 dev-boards and custom-boards).
3. Change the board and press the upload button again
Now you see the compilation starting again for the entire sketch.

Note: this behavior can be seen on the boards expecting a manual push button "boot" is pressed. If I skip pressing the button and the upload correctly fails, the next upload trigger the recompilation instead of starting a new upload.

### Expected behavior

If no changes were made to the sketch, I expect the compilation step is skipped and the upload starting immediately.

### Arduino IDE version

2.0.3

### Operating system

Windows

### Operating system version

10 Pro

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-12-28:
Hi @raffaeler. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/2015.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-12-28 by per1234

raffaeler, 2022-12-28:
Sorry for that. I really looked for the update keyword in the issues, but I didn't see that in the first 8-10 issue **pages**.

weswitt, 2022-12-28:
i opened an issue for this a while ago. issue is still open:

https://github.com/arduino/arduino-cli/issues/1996

per1234, 2022-12-28:
>Sorry for that. I really looked for the update keyword in the issues, but I didn't see that in the first 8-10 issue pages.

No worries @raffaeler. Thanks for taking the time to look for existing issues.

The reason you didn't see it is because it is in a different repository. The codebase hosted here in the `arduino/arduino-ide` repository mostly implements the GUI of Arduino IDE. The operations such as compiling, uploading, library and boards management are all handled by a tool named [**Arduino CLI**](https://arduino.github.io/arduino-cli/latest/). So even though you might be seeing this behavior when using Arduino IDE, the code that implements it is in the codebase in the separate [`arduino/arduino-cli` repository](https://github.com/arduino/arduino-cli).

In order to effectively manage the the development work, we must track the bug reports and feature requests in the repository that contains the relevant codebase. This can sometimes make it difficult for users without the understanding of the low level project architecture to find issues.

raffaeler, 2022-12-28:
@per1234 got it. Maybe you can make it clearer in the template issue.
I specifically looked for the new IDE repository to avoid confusion, but I found no clue in the template description.

per1234, 2022-12-28:
This is an excellent suggestion!

In order to avoid excessive verbosity in the issue template itself (there are additional related projects such as [`arduino/arduino-serial-plotter-webapp`](https://github.com/arduino/arduino-serial-plotter-webapp) that should also be linked to), I would put the information in the [**Issue Report Guide**](https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/issues.md#issue-report-guide) which is linked to from [the "chooser" page](https://github.com/arduino/arduino-ide/issues/new/choose) GitHub presents when creating an issue and also from the [**Contributor Guide**](https://github.com/arduino/arduino-ide/blob/main/docs/CONTRIBUTING.md) that is linked there and in GitHub's issue composer page.