arduino/arduino-ide #1791: Two tabs open for same file and won't close

ChariseWalraven, 2022-12-29:
### Describe the problem

I included a library via the menu and somehow this resulted in another tab opening for the same file. When I try to delete the file nothing happens. I've tried to move the sketch to some other folder and then open and that seems to work (after it renames the parent folder), but when I move it back to the original folder the new tab reappears.

I'm on Ubuntu 22.04.1 LTS and the latest stable version of the IDE shown on the website at the time of writing (2.0.3). I also tested on the latest nightly and it occurs there too.

If you need log files or something, please let me know.

[image]

### To reproduce

I tried reproducing and couldn't. But the issue is still there, even when I open the project on the nightly build.

### Expected behavior

Delete closes the tab. Or closing and reopening the ediitor...

### Arduino IDE version

2.04-nightly-20221229

### Operating system

Linux

### Operating system version

22.04.1 LTS

### Additional context

The fact that this occurs accross builds and after moving things around makes me think this has something to do with some kind of cache or (sketch?) settings file somewhere, but I don't know for sure.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

roboter-basteln, 2022-12-29:
I have a similar issue, but I can't recall having dealt with libraries.
My details see https://arduino.stackexchange.com/questions/91777/how-to-get-rid-of-the-second-tab-in-arduino-ide-2

per1234, 2023-01-03:
Thanks for your reports. Unfortunately, I have not been able to reproduce this fault via use patterns resembling those described here and in the associated reports.

I am aware of two use patterns that will result in a duplicate tabs fault:

## Renaming a sketch file in a manner that only changes the case of the filename

Tracked at https://github.com/arduino/arduino-ide/issues/482

I'm certain that bug is unrelated to the one reported here for the following reasons

- The tabs in this report (and the additional reports) do not show a difference in filename case.
- #482 has existed at least since 2.0.0-beta.10, whereas all reports of this issue are from those using 2.0.3 or newer.

## Opening sketch in a very old version of Arduino IDE 2.x

1. Start any recent version of Arduino IDE.
1. Open a sketch.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start [Arduino IDE 2.0.0-beta.11](https://github.com/arduino/arduino-ide/releases/tag/2.0.0-beta.11).
   **ⓘ** The fault does not occur if you use 2.0.0-beta.12 or newer for this step. I did not bisect it down to the specific newest commit that can be used for this step and still produce the fault.
1. Select **File > Quit** from the Arduino IDE menus.
1. Start any recent version of Arduino IDE.

🐛 The editor will contain duplicate sketch tabs for the sketch you opened previously.

However, I'm not convinced this is the cause of the faults reported here. The reasons:

- The procedure described above produces the fault when using any recent version of Arduino IDE, whereas all reports of this issue are from those using 2.0.3 or newer.
- There are reports of the fault occurring even with a new sketch:
  - https://forum.arduino.cc/t/ide-2-0-3-double-tabs/1066585/1
  - https://forum.arduino.cc/t/ide-2-0-3-double-tabs/1066585/2

## Workaround

The following procedure restores a sketch which is exhibiting the duplicate tabs fault as produced by either of the procedures described above:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**View: Reset Workbench Layout**" command from the menu.

Since I am not able to reproduce the fault via normal usage of Arduino IDE as reported in this issue, I was unable to verify this workaround works under those conditions.

jrobert-gh, 2023-01-04:
I am seeing something similar with 'arduino-ide_nightly-20230104_macOS_64bit.dmg' on macOS Monterey: Selecting certain file tabs duplicates that tab. This behavior has been around earliest Arduino IDE v2.x revisions I've used.

I can delete either of the duplicates by clicking its 'x' in the Open Editors list, but only if the Open Editors list had been open when the duplicate was created. If the list wasn't open and I open it after the duplicate tab is created, the duplicate tab doesn't appear in the list and I can't delete it.

My code files consist of an  .ino file, two .h files, and three .cpp files. Tab duplication seems to happen consistently with two of the .cpp files but not with any of the other four files. The three .cpp files are very similar to one another, having been written from the same template.

The first check-in of the .ino file was March 9, 2014; the project has been through nearly 9 years of Arduino IDE versions (plus Gnu Emacs used about equally).

There are non-code files in the same directory as well:
.uxf - drawing files
.ods - spreadsheet
.pdf
.txt, .lst - text files
.hex - Arduino loader files
.h~, .cpp~, .ino~ - backup files left by Emacs

I don't know what, if any, of the above details will be of any help, but I hope something will.

FernandoGarcia, 2023-01-25:
The same here: 
https://forum.arduino.cc/t/ide-restarts-during-a-start/1081761/3

GaryInSanDiego, 2023-02-01:
I repeatedly had the problem where the sketch file would be open in two tabs.
This is in Arduino IDE  version 2.0.3

I suspect the problem is the sketch file was stored in a file path that was about 165 chars long.  The IDE may not allow for path strings to be that long.

I moved the sketch folder for this sketch file to another location, and the file now opens up in a single tab.

FernandoGarcia, 2023-02-02:
I don't think that the problem is the path since I have the same problem with default file.

[image]

per1234, 2023-02-02:
>I suspect the problem is the sketch file was stored in a file path that was about 165 chars long. The IDE may not allow for path strings to be that long.

Thanks for sharing this information @GaryInSanDiego! I just checked with a sketch in a path 247 characters long and unfortunately still couldn't reproduce the fault.

>I moved the sketch folder for this sketch file to another location, and the file now opens up in a single tab.

An alternative explanation for this is that Arduino IDE stores data about the UI state associated with a given sketch project separately from the sketch files. That data is keyed to the sketch path. So if the bad duplicate tabs state is stored for the sketch at one path, when you move the sketch, you also disassociate it from that previous data and the IDE instead creates a new data entry for the project at that new location.

GaryInSanDiego, 2023-02-02:
I agree the length of the path is not the cause of the problem.

The two tab problem appears to be fixed by moving the sketch folder to another folder.  

After moving the folder, I opened Arduino IDE ver 2.0.3, and it opened with two new blank sketches with todays date (ie. sketch_feb2a.ino just like that pictured above).

I then opened the sketch file that I just moved, and it opened in a single tab.

The sketch file that I was working with was developed in Oct of 2021, and since then I have uninstalled and reinstalled both versions of the Arduino IDE (versions 1.8.57.0 and 2.0.3).  I think they are now installed in C:\Program Files\WindowsApps on my Windows 10 computer.

GaryInSanDiego, 2023-02-03:
I can reproduce the two tabs problem with an old sketch file called RGB_LED.ino (in the ELEGOO starter kit).

I compiled this using Arduino IDE version 1.* back on 8/20/2021.  

Yesterday, I opened this RGB_LED.ino sketch using Arduino IDE version 2.0.3, and the IDE displays two tabs for the same sketch file. I clicked File/Save As, and saved the sketch to another directory with the name myRGB_LED.ino.  The two tab problem persisted in this newly saved version.

I closed the IDE and used Windows 10 File Explorer to copy this sketch file to another folder.  I opened this copied sketch file using Arduino IDE 2.0.3 and only one tab was displayed for the sketch file.  I then used the File/Save As feature to save this sketch to a new name called myRGB_LED_ver2.  The problem of having two tabs for the same sketch file came back.

1) I noticed the IDE text editor remembers positions of highlighted text in both tabs.
2) If you move the sketch file to a new folder using Windows 10 file explorer, the two tab problem goes away.
3) If you use File/Save As to save the sketch file to another folder, two tabs for the same sketch file returns.

I suspect this may have something to do with the IDE editor remembering selected text positions in the sketch file.  If there are two tabs, it remembers selected text in both tabs.   I have not found where the IDE editor stores the editing info yet.

I did look at the folders/files in C:\Users\Name\AppData\Local\Temp and saw several folders for Arduino.  I noticed when I compile a sketch, a sketch folder is produced in this Temp folder (with several sub-folders), which shows paths and file names.

sterretjeToo, 2023-02-25:
I've found the following (seemingly) solid workaround.

If the main sketch file (let's call it A.ino in directory A) shows with a duplicate tab, the following steps remove the duplicate tab
1) Create an additional tab with an ino file (let's call it B.ino); the tab can be left empty
2) Reset the workbench layout as described earlier by @per1234 

The duplicate tab for A.ino has disappeared.

3) You can now remove B.ino (if not needed).

After removing B.ino and reseting the workbench layout again, the duplicate tab for A.ino does not appear again.

Posting to provide a workaround as well as that it might be a pointer for developers to the underlaying issue. It's the second time that I used this approach. The first time I was not sure what I exactly did (except for creating the additional ino file).

Note:
The sketch only consisted of a single ino file, no additional tabs for .h or .cpp or .ino

RocketmanJMP, 2023-03-16:
Any progress on this bug?  I tried all of the above and am still getting a duplicate of an example sketch.  I am using win7 and the IDE version 2.0.4.
Moving and renaming the sketch doesn't fix the issue nor does the F1 reset the workbench.  I also tried deleting the C:\Users\xx\AppData\Roaming\arduino-ide and it always comes back double.  The example sketch is IRremoteESP8266: IRsendProntoDemo

kittaakos, 2023-03-20:
There is a PR with the proposed fix. I could not reproduce the bug on my env, but I found the problem and fixed it. Could somebody help verify the change? The PR is https://github.com/arduino/arduino-ide/pull/1969, and the documentation on how to help with the beta build is here https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md. Thank you!

commit e4beb03a (2023-03-28): fix: incorrect editor widget key calculation

to avoid duplicate editor tabs when opening a sketch with no previously
saved workbench layout

Closes #1791
closed 2023-03-28 by kittaakos