arduino/arduino-ide #1792: Set warnings to "default" by default

roboter-basteln, 2022-12-29:
### Describe the request

I'd appreciate if the warning level would be set to "default" after a fresh installation, so that it produces similar warnings as Arduino IDE version 1.

### Describe the current behavior

Currently, the warning level is set to "none" after a fresh install, not producing any warning.

With the code
```cpp
void setup() {
  // put your setup code here, to run once:
  Serial.begin(100000000000000000000000000000000000000000000000);
}
```
Arduino IDE 1.8.18 gives the warning

```none
...\HelloWorld.ino:3:16: warning: integer constant is too large for its type
   Serial.begin(100000000000000000000000000000000000000000000000);
                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
...\HelloWorld.ino: In function 'void setup()':
...\HelloWorld.ino:3:64: warning: large integer implicitly truncated to unsigned type [-Woverflow]
   Serial.begin(100000000000000000000000000000000000000000000000);

```
when set to warning level "none", while Arduino IDE 2.0.4 does not give any warnings.

### Arduino IDE version

2.0.4-122229

### Operating system

Windows

### Operating system version

Windows 10 22H2

### Additional context

I am working with children at the age of 10, who like experimenting much. It's not uncommon to see such high baud rates. Children love huge numbers. I'd prefer if they get some warnings when they compile their code.
Probably not much different for adults. They should get some warnings, too. You don't want to rely on undefined behavior.

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-12-30:
Thanks for your suggestion @roboter-basteln.

>Arduino IDE 1.8.18 gives the warning

This is actually a bug in Arduino IDE 1.x. After a fresh installation, the preferences are uninitialized. This state causes no warning flags to be added to the compilation commands. You can verify this by performing the following procedure:

1. Select **File > Preferences** from the Arduino IDE menus.
   **ⓘ** Note that the default value of the "**Compiler warnings**" preference is "**None**".
1. Without changing any of the preferences (leaving them in their default states), click the "**OK**" button.
1. Select **Tools > Board > Arduino AVR Boards > Arduino Uno** from the Arduino IDE menus.
   **ⓘ** This is only an arbitrary choice intended to ensure we run the compilation under the same conditions, since the author of each boards platform can decide which flags should be set at any "**Compiler warnings**" preference level.
1. Compile your demonstration sketch.

You will now see that no warnings are printed, as would be expected with a "**Compiler warnings**" preference setting of "**None**". This is because the `-w` flag was added to the compilation commands, as intended by the author of the boards platform:

<https://github.com/arduino/ArduinoCore-avr/blob/1.8.6/platform.txt#L15>

```text
compiler.warning_flags.none=-w
```

---

So we can not use Arduino IDE 1.x's behavior as an argument in favor of your proposal, but that does not invalidate the proposal.

There was a previous proposal for a more extreme change to the default setting of this preference to "**All**": https://github.com/arduino/arduino-ide/issues/1630. I think it is reasonable to maintain two separate proposals even though they are closely related, but I would recommend anyone evaluating this proposal to also check the very relevant discussion at https://github.com/arduino/arduino-ide/issues/1630

Also related (but to be evaluated and implemented independently): https://github.com/arduino/arduino-cli/issues/684

roboter-basteln, 2022-12-30:
@per1234 Thank you for the detailed feedback. I appreciate that someone cares. 
While I was writing the issue, I also thought "It's a bit odd that I get warnings at warning level 'none'". But I didn't want to open an issue for Arduino IDE 1 any more.

per1234, 2024-01-01:
Closing as duplicate of https://github.com/arduino/arduino-ide/issues/1630

closed 2024-01-01 by per1234