arduino/arduino-ide #1793: Real time diagnostics: Unexpected character

roboter-basteln, 2022-12-31:
### Describe the problem

The real time diagnostics reports "Unexpected character", but the code compiles with all warnings enabled.

[image]

### To reproduce

```cpp
void setup() {
  Serial.begin(9600);
}

void loop() {
  Serial.println("😁😒😍🤣😂😊👀✔🤦‍♂️🤦‍♀️🤦‍♀️💕😘👌❤");
}
```

### Expected behavior

Real time diagnostics does not give more errors than compilation.

### Arduino IDE version

2.0.4-nightly-20221229

### Operating system

Windows

### Operating system version

Windows 10 22H2

### Additional context

I am working with 10 year old children. Using smileys is something common.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-01-01:
Thanks for your report @roboter-basteln.

Something I would like to get a clarification on: by default, diagnostics are disabled in Arduino IDE. Did you enable them via the "**Arduino › Language: Real Time Diagnostics**" (`arduino.language.realTimeDiagnostics`) setting in the [advanced settings](https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings)?

---

I did manage to reproduce the fault once using the sketch you provided, but then it went away again and I was never able to reproduce it again despite quite a bit of effort. So it seems either the bug only occurs under some specific set of conditions, or else it is intermittent. @roboter-basteln have you also found the fault to occur inconsistently? Have you used any sort of specific procedure to reproduce the fault beyond what you shared in your report?

---

>I am working with 10 year old children.

The "**Arduino › Language: Real Time Diagnostics**" setting was intentionally disabled by default after the Arduino IDE's [library discovery](https://arduino.github.io/arduino-cli/latest/sketch-build-process/#dependency-resolution) strategy was changed to a "deferred" approach, where discovery is only triggered on compilation. The change was done in order to reduce the much complained about high resource consumption associated with the language server features such as "diagnostics", but there was an unfortunate side effect of a significant increase in the number of false positive "problems" reported by the language server for references to a library API between the time the `#include` directive was added for the header file and when the next library discovery process is triggered.

Even though I think the diagnostics can still be valuable for an experienced user who is able to easily identify such false positives, you might give some careful consideration as to whether it is appropriate to enable the "diagnostics" feature when less experienced users are using the IDE.

roboter-basteln, 2023-01-05:
Yes, I enabled real time diagnostics. That way, children get feedback faster.
Instead of typing 10 lines and then fixing 10 bugs when compiling, they can now type 1 line and fix 1 bug based on real time diagnostics. That's much easier for them to understand.

---
> have you also found the fault to occur inconsistently?
Not yet. 

The tooltip complains about U+1F612 ("unamused face") and U+1F923 ("ROFL").
When I deleted these 2 characters, the error message was still the same. 
It didn't even go away when I deleted all code:
[image]
If that's what you call inconsistent, I agree.
From that behavior, I would guess it has something to do with caching.

---
We haven't used libraries with the new IDE yet, so I can't tell whether false positives increase in that case.