arduino/arduino-ide #1795: Temporary files are not cleaned up

per1234, 2023-01-02:
### Describe the problem

The Arduino IDE application creates some temporary folders:

- Name format: `arduino-ide2-0694b57a37dbff387afae61891204d75/`
  - One folder created for each IDE window that is opened.
- Name format: `.arduinoIDE-unsaved202301-12372-u0dvfq.rtr7i/`
  - One folder created for each unsaved sketch
- Name: [`github-remote/`](https://github.com/eclipse-theia/theia/blob/v1.31.1/packages/plugin-ext/src/main/node/plugin-github-resolver.ts#L39)
- Name: [`http-remote/`](https://github.com/eclipse-theia/theia/blob/v1.31.1/packages/plugin-ext/src/main/node/plugin-http-resolver.ts#L37)
- Name: [`theia_upload/`](https://github.com/eclipse-theia/theia/blob/v1.31.1/packages/filesystem/src/node/node-file-upload-service.ts#L55)

These persist even after the user exits the IDE.

In the case of the first two on the list above, the folders accumulate over time, which may eventually cause a performance impact.

### To reproduce

1. Take note of the contents of your temporary folder.
   - Windows:
     ```text
     C:\Users\<user name>\AppData\Local\Temp
     ```
   - Linux:
     ```text
     /tmp
     ```
   - macOS
     ```text
     $TMPDIR
     ```
1. Start Arduino IDE.
1. Select **File > New** from the Arduino IDE menus.
   **ⓘ** This is treated as an unsaved sketch.
1. Select **File > Quit** from the Arduino IDE menus.
1. Examine the contents of your temporary folder.

🐛 An additional folder is present for each of the IDE windows you opened.
🐛 An additional folder is present for each of the unsaved sketches you opened.
🐛 The `github-remote/`, `http-remote/`, and `theia_upload/` temporary folders Arduino IDE creates are still present

### Expected behavior

Temporary files are cleaned up on exit.

### Arduino IDE version

76f9f63

### Operating system

- Windows
- Linux
- macOS

### Operating system version

- Windows 11
- Ubuntu 22.04
- macOS Ventura

### Additional context

This is similar to https://github.com/arduino/arduino-ide/issues/1779. However, the two are distinct in that https://github.com/arduino/arduino-ide/issues/1779 allows for easier management of the temporary files during runtime (e.g., configuring antivirus software exclusions) and manual cleanup of files leftover when the application is not able to perform an automatic cleanup (e.g., power outage, crash).

---

Some operating systems may eventually do a clean-up on the temporary folder, but this is not guaranteed. It is better for the application to take care of clean up of any temporary files once they are no longer needed.

---

Additional temporary files are generated by tools used by Arduino IDE (e.g., [**Arduino CLI**](https://github.com/arduino/arduino-cli), [**arduino-fwuploader**](https://github.com/arduino/arduino-fwuploader), [**Arduino Language Server**](https://github.com/arduino/arduino-language-server), [**clangd**](https://clangd.llvm.org/)). Cleanup of those file should be the responsibility of the individual tool, not Arduino IDE, so cleanup of those files is out of scope for this issue.

---

Additional reports:

- https://forum.arduino.cc/t/cleanup-of-tmp-seems-inop-kubuntu-20-04/1072075
- https://forum.arduino.cc/t/windows-10-ide-2-0-2-slow-start/1061136/4
- https://forum.arduino.cc/t/rc9-2-temporary-directories/1022812
- https://forum.arduino.cc/t/temporary-files-from-the-arduino-ide/1067309

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

gashtaan, 2024-01-02:
The ticket is year old and there is still massive temporary folder pollution in recent Arduino IDE.

I'm using the IDE very sporadically these days and nevertheless I just had to manually clean hundred files and directories like these:
```
arduino-ide2-XXX
arduino-language-serverXXX
gdb-server-console-XXX.log
system-includes-XXX.clangd
```

dlarue, 2024-01-02:
I run with 2 scripts for Arduino 2.x IDE, one named arduino2 which does not cleanup afterwards and is used when I'm building a few things or running debugging sessions so I might restart the IDE. Then I have arduino22 which is one which calls a tmp-cleanup script after the ide finishes and that does all the /tmp file and directory deletions. Or, after I'm finished with the debugging/etc using arduino2 script I just run the tmp-cleanup script directly.

Kinda a PIA but eased by just needing to run the script isn't too painful.

ednieuw, 2024-06-18:
The Temp\arduino folder should be cleaned regularly. It is now 600 Mb large and full with 0 bytes folders.