arduino/arduino-ide #1796: No Ports Recognized with IDE 2.0.3 ARM64 build (Apple M1) on Mac Ventura

jeffmikels, 2023-01-04:
### Describe the problem

Using the 2.0.3 build for native Apple Silicon works just fine on Monterey, but on Ventura no ports are recognized.

The workaround is to use 2.0.2 under Rosetta. I have not tested the 2.0.3 Intel build, so I don't know if the problem arose with 2.0.3 in general or with the Apple Silicon build specifically.

### To reproduce

- Install 2.0.3 ARM64 build on Mac Ventura
- Run it
- Plug in an Arduino Board
- No ports recognized in the IDE
- even though `ls /dev/cu.*` will correctly show the usb serial device

### Expected behavior

I would expect the IDE to discover the board or at least recognize the existence of the serial port.

### Arduino IDE version

2.0.3 ARM64

### Operating system

macOS

### Operating system version

Ventura (13.1)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-01-04:
Hi @jeffmikels. Thanks for your report. I am also using the Apple Silicon build of Arduino IDE 2.0.3 with macOS Ventura and the ports of my boards are shown in Arduino IDE, so the problem is not universal.

I think the best way to proceed with the investigation is to take a look at the logs generated by Arduino IDE to see whether there is a clue there. You can access the logs by running Arduino IDE from the command line.

Please do this:

1. Open your **Applications** folder.
1. Open **Utilities**.
1. Double-click on "**Terminal**".
1. Type the following command in the **Terminal** window:
   ```text
   "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
   ```
1. Press **<kbd>Enter</kbd>**
1. Wait for Arduino IDE to finish starting.
1. Switch back to the **Terminal** window.
1. Press **<kbd>⌘</kbd>+<kbd>A</kbd>**.
   This will select all the text in the **Terminal** window.
1. Press **<kbd>⌘</kbd>+<kbd>C</kbd>**.
   This will [copy](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the selected text to the clipboard.
1. Click on the "**Leave a comment**" field here on this GitHub issue.
1. Press **<kbd>⌘</kbd>+<kbd>V</kbd>**.
   This will [paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the output into the code block.
1. Click the <kbd>**Comment**</kbd> button to make the reply.

---

Please let me know if you have any questions or problems while following those instructions.

SauceCrusty, 2023-01-06:
I am also having this issue with my Arduino R3 Clone. It works fin in Windows 11 but on MAC Ventura on my M1 it doesn't show any ports. However it does show a serial port in System Report on the MAC when its plugged in.

nickdesaulniers, 2023-01-15:
@jeffmikels @SauceCrusty try installing rosetta? See also #1818.

PaulStoffregen, 2023-01-15:
FWIW, can not reproduce this problem with IDE 2.0.2 with AVR platform 1.8.3 (same as reported in this issue) with MacOS Ventura running on M1 Ultra, uploading to a genuine Arduino Leonardo board.

[image]

PaulStoffregen, 2023-01-15:
Same test works fine with IDE 2.0.3 (x86 version running with Rosetta)

[image]

PaulStoffregen, 2023-01-15:
And for completeness, also works fine with IDE 2.0.3 M1 version, this time with AVR platform 1.8.6 installed.

[image]

SauceCrusty, 2023-01-15:
Not sure why or how but when I tried again I was able to see my ports and upload my sketch.

per1234, 2023-01-18:
Closing due to lack of response to my question from 2 weeks ago, and because @SauceCrusty is no longer able to reproduce the fault.

@jeffmikels if you will provide the requested information by commenting here, I'll be happy to re-open this issue.

closed 2023-01-18 by per1234

jeffmikels, 2023-01-18:
I don't have access to a Ventura machine right now. 😞

per1234, 2023-01-18:
OK, well if you are able to reproduce the issue again, please do comment here. I'll still be monitoring the thread.

The hypothesis offered at https://github.com/arduino/arduino-ide/issues/1796#issuecomment-1383038469 is that the [**serial-discovery**](https://github.com/arduino/serial-discovery) used by Arduino IDE to get a list of the serial ports available on your computer failed to start due to the **Rosetta 2** software not being installed on your Apple Silicon-based computer. **serial-discovery** is built for x86 hosts, which is not a problem when **Rosetta 2** is installed, but a no go without Rosetta. In this case, we would expect to see a "`bad CPU type in executable`" error in the logs you see when starting Arduino IDE from the command line.

jhormigo, 2023-04-09:
I have a Ventura and exactly the same issue. Can't see the ports. I paste below what @per1234 requested:

`Last login: Sun Apr  9 20:33:04 on ttys006
(base) cy@mba13m2 ~ % cd ~/Applications/ArduinoCreateAgent
cd: no such file or directory: /Users/cy/Applications/ArduinoCreateAgent
(base) cy@mba13m2 ~ % cd ~/Applications/ArduinoCreateAgen
Session Contents Restored on 9 Apr 2023 at 21:31

Last login: Sun Apr  9 21:31:09 on console
(base) cy@mba13m2 ~ % ls /dev/cu.*
/dev/cu.Bluetooth-Incoming-Port
(base) cy@mba13m2 ~ % "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
Arduino IDE 2.0.4
Starting backend process. PID: 2266
Using browser-only version of superagent in non-browser environment
Configuration directory URI: 'file:///Users/cy/.theia'
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
2023-04-09T19:49:18.896Z root INFO Backend Object.initialize: 1.9 ms [Finished 0.355 s after backend start]
2023-04-09T19:49:18.896Z root INFO Backend Object.configure: 1.5 ms [Finished 0.355 s after backend start]
2023-04-09T19:49:18.896Z root INFO Backend MessagingContribution.onStart: 0.7 ms [Finished 0.355 s after backend start]
2023-04-09T19:49:18.897Z root INFO Backend OSBackendApplicationContribution.configure: 0.2 ms [Finished 0.371 s after backend start]
2023-04-09T19:49:18.897Z root INFO Backend DefaultWorkspaceServer.onStart: 0.1 ms [Finished 0.371 s after backend start]
2023-04-09T19:49:18.897Z root INFO Backend PluginLocalizationBackendContribution.configure: 0.1 ms [Finished 0.371 s after backend start]
2023-04-09T19:49:18.898Z root INFO Backend PluginLocalizationBackendContribution.initialize: 16.0 ms [Finished 0.371 s after backend start]
2023-04-09T19:49:18.898Z root INFO Backend TaskBackendApplicationContribution.onStart: 0.1 ms [Finished 0.371 s after backend start]
2023-04-09T19:49:18.898Z root INFO Backend ElectronTokenBackendContribution.configure: 0.5 ms [Finished 0.372 s after backend start]
2023-04-09T19:49:18.898Z config INFO >>> Initializing CLI configuration...
2023-04-09T19:49:18.898Z root INFO Backend PluginDeployerContribution.initialize: 0.2 ms [Finished 0.372 s after backend start]
2023-04-09T19:49:18.898Z root INFO Backend ConfigServiceImpl.onStart: 0.4 ms [Finished 0.372 s after backend start]
2023-04-09T19:49:18.899Z config INFO Loading CLI configuration from /Users/cy/.arduinoIDE/arduino-cli.yaml...
2023-04-09T19:49:18.899Z root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.2 ms [Finished 0.373 s after backend start]
2023-04-09T19:49:18.899Z root INFO Backend ArduinoDaemonImpl.onStart: 0.5 ms [Finished 0.373 s after backend start]
2023-04-09T19:49:18.899Z root INFO Backend NodeFileUploadService.configure: 0.8 ms [Finished 0.373 s after backend start]
2023-04-09T19:49:18.899Z daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
2023-04-09T19:49:18.899Z discovery-log INFO start
2023-04-09T19:49:18.899Z discovery-log INFO start new deferred
2023-04-09T19:49:18.899Z root INFO Backend HostedPluginLocalizationService.initialize: 0.2 ms [Finished 0.374 s after backend start]
2023-04-09T19:49:18.899Z root INFO Backend BoardDiscovery.onStart: 0.5 ms [Finished 0.374 s after backend start]
2023-04-09T19:49:18.899Z root INFO Backend FileDownloadEndpoint.configure: 0.4 ms [Finished 0.374 s after backend start]
2023-04-09T19:49:18.899Z root INFO Backend AuthenticationServiceImpl.onStart: 0.1 ms [Finished 0.375 s after backend start]
2023-04-09T19:49:18.899Z root INFO Backend PluginApiContribution.configure: 0.1 ms [Finished 0.375 s after backend start]
2023-04-09T19:49:18.900Z root INFO Backend HostedPluginReader.configure: 0.2 ms [Finished 0.375 s after backend start]
2023-04-09T19:49:18.900Z root INFO Backend HostedPluginReader.configure: 0.1 ms [Finished 0.375 s after backend start]
2023-04-09T19:49:18.900Z root INFO Backend PlotterBackendContribution.configure: 0.2 ms [Finished 0.375 s after backend start]
2023-04-09T19:49:18.901Z root INFO Theia app listening on http://localhost:53577.
2023-04-09T19:49:18.901Z root INFO Finished starting backend application: 1.6 ms [Finished 0.376 s after backend start]
2023-04-09T19:49:18.903Z root WARN The local plugin referenced by local-dir:/Users/cy/.arduinoIDE/plugins does not exist.
2023-04-09T19:49:18.903Z root WARN The local plugin referenced by local-dir:/Users/cy/.arduinoIDE/extensions does not exist.
2023-04-09T19:49:18.903Z root WARN The local plugin referenced by local-dir:/Users/cy/.arduinoProIDE/plugins does not exist.
2023-04-09T19:49:18.920Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-04-09T19:49:18.920Z root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@1.5.1" with engines:
2023-04-09T19:49:18.920Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-04-09T19:49:18.920Z root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.7" with engines:
2023-04-09T19:49:18.920Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-04-09T19:49:18.920Z root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
2023-04-09T19:49:18.920Z root INFO PluginTheiaDirectoryHandler: accepting plugin with path
2023-04-09T19:49:18.920Z root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
[... cut ...]
2023-04-09T19:49:22.467Z daemon INFO INFO[0003] Required tool                                 tool="arduino:arm-none-eabi-gcc@7-2017q4"
INFO[0003] Required tool                                 tool="arduino:bossac@1.9.1-arduino2"
INFO[0003] Required tool                                 tool="arduino:dfu-util@0.10.0-arduino1"

2023-04-09T19:49:22.467Z daemon INFO INFO[0003] Required tool                                 tool="arduino:openocd@0.11.0-arduino2"
INFO[0003] Required tool                                 tool="arduino:rp2040tools@1.0.6"

`

grandinquisitor, 2023-07-15:
I found that I was able to work around this issue only by making sure my board was plugged in BEFORE starting the IDE. If I plugged it in after the IDE was started it never found it.