arduino/arduino-ide #1797: I can't open files in WSL

119311, 2023-01-04:
### Describe the problem

# Cannot open files in WSL

I am using Windows 10.

I can't open sketches either from explorer or from arduino IDE.

### To reproduce

Open Arduino IDE and select Open... then choose sketch in WSL. 

My environment is as follows.

---
Arduino IDE
	Version: 2.0.4-nightly-20230104
	I also tried with 2.0.3

Windows and WSL
	WSL Version: 1.0.3.0
	Kernel Version: 5.15.79.1
	WSLg Version: 1.0.47
	MSRDC Version: 1.2.3575
	Direct3D Version: 1.606.4
	DXCore Version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
	Windows Version 10.0.19045.2364

WSL	distro

	OS:	Ubuntu	22.04.1	LTS	on	Windows	10	x86_64
	Kernel:	5.15.79.1-microsoft-standard-WSL2
	
	Packages:	1748	(dpkg),	4	(snap)
	Shell:	zsh	5.8.1
	Theme:	Adwaita	[GTK3]
	Icons:	Adwaita	[GTK3]
	Terminal:	Windows	Terminal
	CPU:	AMD	Ryzen	5	4500U	with	Radeon	Graphics	(6)	@	2.370GHz
	GPU:	e775:00:00.0	Microsoft	Corporation	Device	008e
	Memory:	1747MiB	/	7946MiB

### Expected behavior

Sketch in WSL will not be opened, but a new sketch will be opened.

### Arduino IDE version

2.0.4-nightly-20230104

### Operating system

Windows

### Operating system version

10.0.19045.2364

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-01-04:
Hi @119311 please tell us the full path of one of the sketches that won't open.

119311, 2023-01-04:
Hi @per1234 .

It looks like you didn't have enough information
sorry.

sketch's path is like "\\wsl$\Ubuntu\home\adobe\report\HCI\w1\src\ReadPushSwitch\ReadPushSwitch.ino".
and "\\wsl$\Ubuntu\home\adobe\report\HCI\w1\src\SerialCount\SerialCount.ino" , "\\wsl$\Ubuntu\home\adobe\report\HCI\w1\src\SerialCount2\SerialCount2.ino"

per1234, 2023-01-05:
Is is `\wsl$\`... or `\\wsl$\`...? (note the first starts with a single `\`, while the second starts with two `\`)

I ask because there is a known bug that produces this fault when attempting to open sketches from any [UNC path](https://en.wikipedia.org/wiki/Path_(computing)#Universal_Naming_Convention) like `\\wsl$\`...

Unfortunately I don't have any experience with WSL (I use a standard Linux machine when I want Linux and [**Git Bash**](https://gitforwindows.org/#bash) to get a sufficiently Linux-ish experience from my Windows machine), so I apologize if I am a bit slow to understand the details of the report.

119311, 2023-01-05:
The path started with "//".
Oh, my point was that I could not open the UNC path.

I see, so there is a known bug.
If so, would I close this issue?

per1234, 2023-01-05:
Thanks for the clarification.

> would I close this issue?

The bug is actually in the [**Eclipse Theia** IDE framework](https://theia-ide.org/) Arduino IDE is built on. I submitted a report to that project's repository:

https://github.com/eclipse-theia/theia/issues/11997

But I think it is reasonable for us to document the bug in this project's issue tracker as well so I think we should leave this issue open until the bug is resolved by updating the Theia version used by Arduino IDE to one that does not have the bug (we have to wait for a fix and a release before that will be possible). Even though the bug is not in the codebase hosted in this repository, that fact is not at all obvious so affected users will come here to look for an existing report rather than the Theia repository.

119311, 2023-01-05:
Ah, so the arduino IDE was based on theia, not codium!
I was wondering why I could open it with the codium.
Thank you for your response.

giano2002, 2023-04-05:
Hi, 
I'm also having trouble opening .ino files that are saved on the NAS. **Using IDE 2.0.4, when I try to open an .ino file (example: "_basics.ino_"), saved in a folder on the NAS, the empty file "_sketch_apr5a.ino_" opens, not the one I have chosen. If I use, instead , IDE 1.8.19.0, the file opens normally.** 
I am using Windows 11 Home, and the network path is as follows: "\\\NAS_1\home\IL MIO DRIVE\ARDUINO_SKETCH\basics\basics.ino".
Following some suggestions, I solved the problem after mapping the network folder (\\NAS_1\home\MY DRIVE) (Z:). But I think this is a temporary solution, waiting for the programmers to finally fix the problem.

I attach my system information:
[System Info.txt](https://github.com/arduino/arduino-ide/files/11156144/System.Info.txt)
Thanks

119311, 2023-04-05:
I don't know all the details...
The problem of not being able to read files on this network seems to depend on theia.
https://github.com/eclipse-theia/theia/issues/11997

Silver-Fang, 2024-06-12:
Isn't VSCode also based on Theia? Why doesn't it have this problem? Can you refer to VSCode's solution for Arduino IDE?

Dynteq, 2025-02-04:
I can confirm this still exists in 2.3.4
Trying to open the .ino by navigating to the file location through windows "quick access" will not work and result in a new default file opening.
Workaround; navigating to the file through the mapped drive and manually drilling down to the right directory and then opening with IDE does work.