arduino/arduino-ide #1798: SaveAs looses end of sketch

ednieuw, 2023-01-04:
### Describe the problem

When I develop a program I regularly save the script with a new filename containing the version number.
From the first version of the IDE 2.0 I have had problems with this SaveAS. I therefore often fall back to the IDE 1.8
Or IDE 2.0 saved the sketch in the Arduino sketch root folder or it froze during a save  or lost a part of the sketch. 
Loosing the last part of the sketch is what happened with IDE V2.0.3 again
I attached the last compilations of Version RP2040-WordClockV018b and the damaged file RP2040-WordClockV019.
Before SaveAs I saw that the V018b filename had a lot of, probably, temporary characters after the V018b like RP2040-WordClockV018b344uiy32i4ui3 or something like that
I regularly put the PC in Hibernation or in Sleep then taking a pause for myself

### To reproduce

I have no idea to reproduce this
[Temp.zip](https://github.com/arduino/arduino-ide/files/10344842/Temp.zip)
Attached the ZIP with the last compilation temp files with all the use libraries et cetera
I compile for an Arduino Nano RP2040 Connect with the standard Arduino Board settings.
Autosave  is off because I regularly Save the file manually

arduino-language-server1763127236 contains the last complete V018b INO file before the SaveAs V019
arduino-language-server3607719326 contains the damage V019 INO file

### Expected behavior

Part of the INO file is lost

### Arduino IDE version

2.0.3

### Operating system

Windows

### Operating system version

W11

### Additional context

Still miss my Save button 

I did not test with the nightly built

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-01-04:
Hi @ednieuw. Thanks for your report.

> I attached the last compilations of version `RP2040-WordClockV018b` and the damaged file `RP2040-WordClockV019`.

I think you forgot to attach the "damaged file `RP2040-WordClockV019`"

It is not clear to me why you chose to include the temporary files generated by Arduino IDE instead of the permanent sketch folders. These files are only intended to be used internally by the [**Arduino Language Server**](https://github.com/arduino/arduino-language-server) and [**Arduino Firmware Uploader**](https://arduino.github.io/arduino-fwuploader/latest/), rather than directly by the user. The user is expected to only directly interact with the permanent sketch files produced by Arduino IDE's "Save As" and "Save" operations. Do you use these files directly? If yes, please describe how you use them.

ednieuw, 2023-01-05:
Hi Per1234

 

The attached ZIP file is a kind of history. It contains the used libraries and the script. Because I could not reproduce the error I attached the last compilations as a ‘proof’ that it is happening on my PC. But there is not much for you to do with.
When I do a File->Saveas I see: RP2040-WordClockV019_copy_20230105143508
There is a lot  of timestamp gibberish attached after the original file name. Maybe a separate issue to solve and do not add ‘_copy_20230105143508’.

My problem is that something ends the saving process before it is finished and without a warning. 

 

!!!!  I repeated the problem once while typing this mail. The other 10 times the file saveas’d  correctly.

Attached the original and corrupted file.

Cheers

Ed

per1234, 2023-11-23:
I see we received another report of this bug later, which received more discussion: https://github.com/arduino/arduino-ide/issues/2086

The cause of that bug was identified and fixed by https://github.com/arduino/arduino-ide/pull/2027. I checked the files you provided:

```text
$ stat --printf="%s" arduino-language-server1763127236/sketch/RP2040-WordClockV018b.ino.cpp 

111016

$ stat --printf="%s" arduino-language-server3607719326/sketch/RP2040-WordClockV019.ino.cpp 

70321
```

From this I can see that the symptoms you reported here match very well with those reported in https://github.com/arduino/arduino-ide/issues/2086

So I will close this as fixed. The fix was first introduced in the Arduino IDE 2.2.0 release so you should not encounter this problem when using Arduino IDE versions 2.2.0 or newer. If you are still experiencing this fault, please comment here and I'll investigate further.

Thanks again for reporting this serious bug @ednieuw. I apologize for not having found the time to look into this before now.

closed 2023-11-23 by per1234