arduino/arduino-ide #1799: Allow Output and Serial Monitor to be docked beside each other.

TheConceptBoy, 2023-01-05:


per1234, 2023-01-05:
Hi @TheConceptBoy. Thanks for your suggestion. Arduino IDE previously had this capability and the project managers decided that it did not align with the intended design architecture for Arduino IDE, so it was intentionally removed: https://github.com/arduino/arduino-ide/pull/940

Regardless of whether you agree with that, the proposal specific to the "**Serial Monitor**" will be obviated anyway by the planned reworking of the "**Serial Monitor**" feature as a dedicated window similar to the "**Serial Plotter**" feature: https://github.com/arduino/arduino-ide/issues/289#issuecomment-1070964887

closed 2023-01-05 by per1234

TheConceptBoy, 2023-01-09:
Can they at least prevent the Output window from going into focus on every upload? It's pretty annoying when debugging via Serial monitor while doing rapid sketch uploads to have to keep switching to Serial Monitor as the Output keeps taking focus.

Compared to this, docking side by side seems like a much more reasonable solution.