arduino/arduino-ide #1800: Manually disconnect serial port

falk-stefan, 2023-01-05:
### Describe the request

I want to be able to manually disconnect the serial port connection using a toggle-button.

At the moment I am using a JavaScript client in order to visualize serial data. However, in order to be able to connect via serial connection, I have to close the Arduino IDE entirely. 

It would be nice if we could just close the connection if we want to.

### Describe the current behavior

Afaik there is no way to close the serial connection.

### Arduino IDE version

2.0.3

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2023-01-05:
Hi @falk-stefan. Thanks for your suggestion.

I need to understand how you are using the serial port in Arduino IDE.

Do you have a "**Serial Monitor**" tab in the bottom panel of your IDE?:

[image]

Did you use **Serial Plotter** at any time during the Arduino IDE session?:

[image]

falk-stefan, 2023-01-05:
Hi!

This has nothing to do with the Serial Monitor as such. I just want to open a web browser and connect to the serial port where the Arduino is connected. However, if the Arduino IDE is open, then the board gets connected automatically for programming the Atmega. In my case e.g. the COM3 port is used and connected. If I now want to use my JavaScript client like so:

```javascript
const port = await navigator.serial.requestPort();
// Wait for the serial port to open.
await port.open({ baudRate: 9600 });
```` 

it won't work because Arduino is occupying the port.

```
Uncaught (in promise) DOMException: Failed to execute 'open' on 'SerialPort': Failed to open serial port.
```

The only way around this it seems is to close the IDE.

per1234, 2023-01-05:
@falk-stefan please just give a straightforward answer to the questions I asked:

Is there a "**Serial Monitor**" tab in the bottom panel of the IDE window? If you don't see the bottom panel, you can open it by clicking this icon at the right side of the status bar:

[image]

Did you use Serial Plotter at any time during the Arduino IDE session?

What you need to understand is that Arduino IDE itself does not open the selected serial port other than during the upload process. So we are left with three possible causes of the problem you are experiencing:

- You have Serial Monitor running (which does open the port)
- You have Serial Plotter running (which does open the port) or you used Serial Plotter at some time during the session (there is a known bug that the port is not closed)
- There is a bug in Arduino IDE that is causing it to open the port.

We must determine which of these three things is causing the problem. In order to do that, you must answer my questions.

If it is the first, then you simply need to close Serial Monitor.

If it is the second, then we already have an issue tracking the bug of the port not being closed.

If it is the third, then we must investigate the bug and fix it.

---

You are presenting an [XY problem](https://xyproblem.info/) here. You think the solution (X) is to add a GUI element to close the port. But the real problem (Y) is that the serial port is open in the first place.

In the case of Serial Monitor, we already have that GUI element (the **X** icon on the view tab). In the case of Serial Plotter, we already have that GUI element (the window close control), but it is not working currently. In the case of the IDE itself, there is no point in adding a GUI element because the hypothetical bug should just be fixed instead.

falk-stefan, 2023-01-05:
You were right by suggesting to close the Serial Monitor. I was under the impression that the port is being opened and kept open during the IDE is running. 

Closing the Serial Monitor, as you said, fixes the issue. Thanks.

closed 2023-01-05 by falk-stefan

per1234, 2023-01-05:
You are welcome. I'm glad it is working now.

I think the current UI design does not make the need to close Serial Monitor to free the port obvious to the user. That poor user experience should be resolved as a matter of course when Serial Monitor is moved to a dedicated window: https://github.com/arduino/arduino-ide/issues/289#issuecomment-1070964887

falk-stefan, 2023-01-05:
Got it. Looking forward to upcoming upgrades 👍