arduino/arduino-ide #1801: [Windows 11] uploading continues on and on despite of seemingly finished uploading

lun82jiang, 2023-01-05:
### Describe the problem

I have a dual boot system with windows 10 and windows 11.
In windows 10 I have no problem, but in windows 11, the 
[image]
goes on and on although the verbose mode uploading shows a completion of the job (I assume).
It seems windows 11 is still occupying the USB communication somehow and not releasing it.

### To reproduce

Windows 11 program, try to upload the following program.
#include <SoftwareSerial.h>
SoftwareSerial mySerial(10, 11); // RX, TX

char Mymessage[10]; //Initialized variable to store recieved data
  
  void setup() {
    Serial.begin(9600);
    // Begin the Serial at 9600 Baud
    mySerial.begin(9600);
  }
  
  void loop() {
    mySerial.readBytes(Mymessage,5); //Read the serial data and store in var
    //Serial.print(Mymessage);
    Serial.println(Mymessage); //Print data on Serial Monitor
    delay(1000);
  }

### Expected behavior

[image]
goes on and on, 
verbose output:
Sketch uses 3222 bytes (9%) of program storage space. Maximum is 32256 bytes.
Global variables use 307 bytes (14%) of dynamic memory, leaving 1741 bytes for local variables. Maximum is 2048 bytes.
"C:\Users\wormi\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/bin/avrdude" "-CC:\Users\wormi\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf" -v -V -patmega328p -carduino "-PCOM4" -b115200 -D "-Uflash:w:C:\Users\wormi\AppData\Local\Temp\arduino-sketch-0D4DC7263DFA72C1021B25AD2D7907C6/TestWriter.ino.hex:i"

avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\wormi\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17/etc/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "C:\Users\wormi\AppData\Local\Temp\arduino-sketch-0D4DC7263DFA72C1021B25AD2D7907C6/TestWriter.ino.hex"
avrdude: writing flash (3222 bytes):

Writing | ################################################## | 100% 0.53s

avrdude: 3222 bytes of flash written

avrdude done.  Thank you.

### Arduino IDE version

2.0.3 CLI version 0.29.0

### Operating system

Windows

### Operating system version

Windows 11 Pro version 22H2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-01-06:
Hi @lun82jiang. Thanks for your report.

I would like to learn more about the bug you reported:

---

Does this problem occur every time you attempt an upload on your Windows 11 machine, or is it intermittent (e.g., sometimes the IDE releases from the "Uploading..." state when the upload process finishes, and other times the IDE stays perpetually in the "Uploading..." state)?

---

Please try this experiment:

1. Start Arduino IDE 2.0.3 on your Windows 11 machine.
1. Select **File > New** from the Arduino IDE menus.
1. Connect your Arduino board to your computer.
1. Select the appropriate board and port in Arduino IDE.
1. Select **Sketch > Upload** from the Arduino IDE menus.

Please comment here to let us know whether Arduino IDE remains in the "Uploading..." state even after the upload of that bare minimum sketch finishes?

The purpose of this experiment is to determine whether uploading the specific sketch you provided above is a required condition for the fault to occur, or whether the fault occurs with any arbitrary sketch being uploaded.

lun82jiang, 2023-01-06:
Hi Per1234,
I was able to resolve the problem doing the following:
Reinstalled Arduino IDE
Deleted the AppData\Local\Arduino15

Now it seems I was doing Serial communication on the RX/TX (P00 and P01) on Uno (using SoftwareSerial), and then I started reprogramming the board. This left some kind of state on the Windows 11 system.
I'm not sure if it's going to come back. But if it does, I'll update this thread.

lun82jiang, 2023-01-06:
> Hi @lun82jiang. Thanks for your report.
> 
> I would like to learn more about the bug you reported:
> 
> Does this problem occur every time you attempt an upload on your Windows 11 machine, or is it intermittent (e.g., sometimes the IDE releases from the "Uploading..." state when the upload process finishes, and other times the IDE stays perpetually in the "Uploading..." state)?
> 
> Please try this experiment:
> 
> 1. Start Arduino IDE 2.0.3 on your Windows 11 machine.
> 2. Select **File > New** from the Arduino IDE menus.
> 3. Connect your Arduino board to your computer.
> 4. Select the appropriate board and port in Arduino IDE.
> 5. Select **Sketch > Upload** from the Arduino IDE menus.
> 
> Please comment here to let us know whether Arduino IDE remains in the "Uploading..." state even after the upload of that bare minimum sketch finishes?
> 
> The purpose of this experiment is to determine whether uploading the specific sketch you provided above is a required condition for the fault to occur, or whether the fault occurs with any arbitrary sketch being uploaded. Reinstalling/Restarting IDE didn't help. Even flashing basic blink program would still result in being stuck at Uploading....

When the problem persisted, it was NOT intermittent. Since I had 2 boards, soon both started to have the same problem (I was loading the faulty program on both of them).

willishf, 2023-01-13:
Having similar issues as others regarding the "loading" message not going away after deploying code via mounted drive for a RP2040 pico W. Don't think it is a specific problem to pico W but more related to the code being deployed, starts running, has an issue and creates a race condition for the IDE waiting for a response related to setting up com port. 

The following code is a simple test of trying to get Dallas OneWire protocol working on a Pico W. Deploy the code, loading message doesn't go away, get a windows USB error message notification and requires exiting the IDE and pushing boot sel button via pico power up. 

Deploy blink code and works as expected. If I comment out the code related to sensors.begin() and reading sensor temperature then code is able to deploy and works related to sending back messages in the loop. 

You should be able to deploy the following code on a pico W and get the error. 

I am runnong on Arduino IDE 2.0.3 using earlephilhower board manager  https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json to select for pico W.

The arduino support for pico does not extend to pico W as the pico W hijacked a GPIO line used for the blink code for the Wifi module and then gave it back via external hardware. 

/* Arduino DS18B20 temp sensor tutorial
   More info: http://www.ardumotive.com/how-to-use-the-ds18b20-temperature-sensor-en.html
   Date: 19/6/2015 // www.ardumotive.com */

//Include libraries
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into pin 2 on the Arduino
#define ONE_WIRE_BUS 2
// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);
// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

void setup(void)
{
  Serial.begin(9600); //Begin serial communication
  Serial.println("Arduino Digital Temperature // Serial Monitor Version"); //Print a message
  sensors.begin();
}

void loop(void)
{ 

  // Send the command to get temperatures
  sensors.requestTemperatures();  
  Serial.println("Temperature is: ");
  Serial.println(sensors.getTempFByIndex(0)); // Why "byIndex"? You can have more than one IC on the same bus. 0 refers to the first IC on the wire
  //Update value every 1 sec.
  delay(1000);
}

tigoe, 2023-01-31:
@per1234, I am seeing a similar issue on MacOS Monterey 12.6.3 using IDE 2.0.3. and SAMD boards (Nano 33 IoT, MKR 1010, MKR Zero): 
Symptoms:
* Connected SAMD boards intermittently de-enumerate from the list of ports.

Steps to reproduce:
* Plug in a SAMD board (tested with the models above)
* Load a sketch (tested with WiFiNINA library examples and Adafruit_NeoPixel examples)
*  Type command-U to upload
* Wait. 

Sometimes the program will upload fine. 
Sometimes it will upload, but the serial port will not reappear
Sometimes it will reappear, then disappear when you make any change to the code and upload a second time.

Steps attempted to fix (these do not work consistently):
* remove ~/LIbrary/Adruino15 folder and restart
* re-install SAMD boards
or
* double-tap reset button to put board into bootloader mode
* type command-U

It's a strange problem in that it is inconsistent, yet happens frequently enough to prohibit useful work. My solution has been to back down to IDE 1.8.19, which is not an adequate solution.

Another observation, perhaps unrelated: if I leave IDE2.0 open when closing the lid of my laptop, the OS will occasionally restart while closed. If I make sure to close the IDE2.0, the OS doesn't restart.

commit f1e97d00 (2023-03-31): fix: updated to `electron-builder@23.6.0`

Closes: #1801