arduino/arduino-ide #1806: Serial Monitor did nothing

tyeth, 2023-01-08:
### Describe the problem

Upload worked fine, I was testing the Sensirion SEN55 exampleUsage sketch on an adafruit QTPY ESP32-S2. I also have the adafruit wippersnapper sketch open in a first window, but no serial monitor open there.

I had been messing with CDC mode on and off because of seeing no serial monitor output (closed and opened serial monitor). I uploaded multiple times with no problems but never serial monitor entries, despite reboots/reinserts of device and reselecting ports etc. Only when I opened another example sketch in another window to check the i2c bus (adafruit example), then started the compile and upload, switched back to the SEN55 example window and closed the serial monitor, then starting it in the new window did the new window have a working serial monitor showing the SEN55 sketch output, but was almost immediately closed as the upload started.

Serial monitor should work reliably. Please investigate or offer a debug logging build that users with troubles can run and log with.

### To reproduce

🤷 

### Expected behavior

Serial monitor works in any window assuming only a maximum of one serial monitor is open across all windows/sketches.

### Arduino IDE version

2.0.3

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

tyeth, 2023-01-08:
I have not verified the nightly as I do not know recreation steps, sorry. Hence the ask for a troubleshooting / logging build that records every session just in case. The user accepts this may fill up their disk.

per1234, 2023-01-14:
Hi @tyeth. Thanks for your report

> Hence the ask for a troubleshooting / logging build that records every session just in case.

Arduino IDE automatically generates logs without any need for a special build. The log files are saved to this folder:

```text
C:\Users\<username>\AppData\Roaming\Arduino IDE\
```

(where `<username>` is your Windows username)

Please let us know if you find anything of interest in the logs after you experience the problem next.

tyeth, 2023-01-14:
[2023-01-08_log.log](https://github.com/arduino/arduino-ide/files/10418108/2023-01-08_log.log)
[2023-01-07_log.log](https://github.com/arduino/arduino-ide/files/10418109/2023-01-07_log.log)
The 8th is the day in question, but I include logs from the 7th at 100kb, and the 8th at 1.7mb. Couldn't see the wood for the trees so leaving this here for comparison by future me...

tyeth, 2025-08-26:
Thanks @per1234, I have long forgotten the issue. Closing for now

closed 2025-08-26 by tyeth