arduino/arduino-ide #1813: Could not install Arduino AVR platform: Error: 2 UNKNOWN

AlexisComix, 2023-01-10:
### Describe the problem

When installing the Arduino AVR platform for Arduino Uno, I get the following error notification:
```
Could not install Arduino AVR platform: Error: 2 UNKNOWN: read tcp {my IP:some port}->{another IP}:80: wsarecv: An existing connection was forcibly closed by the remote host.
```
The second IP varies, as well as the port for my IP.
From multiple attempts, the second IP shown seems to always be a Cloudflare IP, through port 80.

The Output window has the following:
```
Downloading packages
arduino:avr-gcc@7.3.0-atmel3.6.1-arduino7
Failed to install platform: arduino:avr.
Error: 2 UNKNOWN: read tcp {my IP:port}->{another IP}:80: wsarecv: A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond.
```
The functions for verifying, uploading, and debugging the program do not work. 

The issue was on the latest stable release as well as the latest nighty release: I have not tried previous versions.

### To reproduce

To reproduce, open a version of the software that has not had the AVR installed yet, plug in an Arduino Uno, and press the install button on the popup to install the AVR.

### Expected behavior

The expected behaviour is the full installation of the AVR for the Arduino Uno, allowing for the Verify, Upload, and Debug functions to work.

### Arduino IDE version

2.0.4-nightly-20230110

### Operating system

Windows

### Operating system version

Windows 11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

AlexisComix, 2023-01-11:
Sorry, turns out this was likely a firewall issue. Will mark as closed.

LukeMoglia, 2023-03-23:
Hi, I'm currently facing the same issue with our firewall. Do you know what it was you had to whitelist/change on the firewall to allow it to work?

Cheers

AlexisComix, 2023-03-23:
Hi, unfortunately I don't own the network as it was at a university, however what lead me to believe it was a firewall issue is that it seemed to work fine when I connected to a mobile data hotspot. While I am very unexperienced in this area yet, I figured maybe the network had some rule to do with CloudFlare IPs and I put it down to that. I'm sorry for not having a definitive fix, I hope you can get the problem sorted soon :)

greenrupture, 2023-08-25:
I have the same issue trying to install Arduino UNO R4 Boards and other board platforms, I received my Arduino UNO R4, but I can't install the core in the Arduino IDE using the Boards Manager.

I've attempted the following troubleshooting steps:

    Disabled the firewall
    Tried using different internet connections (work, home, etc.)
    Restarted my PC
    Reinstalled the Arduino IDE
    Tried using my friend's PCs
    Executed Arduino IDE as an administrator
    Tried Arduino IDE version 1.8xx
    Attempted to install an older version of the core
    Tried installing in different days

I tried with a faster internet connection, and it worked. I think this issue might be a timeout error, but the timeout threshold seems to be too low. Not everybody has access to a fast internet connection, which doesn't align well with the spirit of 'Open Source,' in my opinion.
What can we do to increase the timeout threshold?

michielper, 2024-12-23:
Same problem here, no clue why I cannot install the Arduino IDE on my Ubuntu system. I am on a normal home network that otherwise gives me no issues. Should I enable some special ports?
Error message:
Failed to install platform: 'Arduino AVR Boards:1.8.6'. 2 UNKNOWN: read tcp [2a02:a44d:9f1d:1:7a50:eb86:e87c:3ff9]:47972->[2606:4700::6812:a15]:443: read: connection reset by peer