arduino/arduino-ide #1816: Wayland + Nvidia driver = black screen

Attigliuzzo, 2023-01-12:
### Describe the problem

I'm using Arduino Ide 2.0.3 on 2 pc, on both of them i've installed wayland under Debian testing.

On my old laptop, wayland uses mesa on an Intel discrete graphic card,  the ide starts successfully and i see the full interface.

On the newer one, i use Nvidia driver 510 with wayland, the whole screen app is black. i think it is not well supported under wayland. 
I can use X11 instead.

Anyway from console i get:

> Arduino IDE 2.0.3
> Starting backend process. PID: 4204
> [4181:0112/163131.466177:ERROR:object_proxy.cc(642)] Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.InvalidArgs: Interfaccia «org.freedesktop.portal.FileChooser» inesistente
> [4181:0112/163131.466205:ERROR:select_file_dialog_impl_portal.cc(243)] Failed to read portal version property
> Using browser-only version of superagent in non-browser environment
> Configuration directory URI: 'file:///home/attila/.theia'
> Configuring to accept webviews on '^.+\.webview\..+$' hostname.
> 2023-01-12T15:31:32.211Z root INFO Backend Object.initialize: 3.6 ms [Finished 0.791 s after backend start]
> 2023-01-12T15:31:32.211Z root INFO Backend Object.configure: 3.1 ms [Finished 0.791 s after backend start]
> 2023-01-12T15:31:32.212Z root INFO Backend MessagingContribution.onStart: 1.0 ms [Finished 0.791 s after backend start]
> 2023-01-12T15:31:32.212Z root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 0.814 s after backend start]
> 2023-01-12T15:31:32.212Z root INFO Backend DefaultWorkspaceServer.onStart: 0.2 ms [Finished 0.814 s after backend start]
> .........

### To reproduce

Launch the app image in a wayland environment with proprietary Nvidia driver.

### Expected behavior

An unreadable/inusable interface, better a black screen.

### Arduino IDE version

2.0.3

### Operating system

Linux

### Operating system version

Debian testing "Bookworm"

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-01-17:
Here, people say the hardware acceleration of the electron app should be disabled: https://github.com/signalapp/Signal-Desktop/issues/5361. The app is different but the env is the same.

Possible related: https://github.com/arduino/arduino-ide/issues/142

Mahiuha, 2023-01-19:
It appears that the issue is related to a problem with the file chooser dialog in the Arduino IDE when running on Wayland with the Nvidia driver. The error message "Failed to call method: org.freedesktop.DBus.Properties.Get: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.InvalidArgs: Interfaccia «org.freedesktop.portal.FileChooser» inesistente" suggests that the required file chooser interface is not present.

A possible solution would be to try running the Arduino IDE with the Xorg display server instead of Wayland, as it is known to have better compatibility with certain applications. Another solution would be to try using a different version of the Nvidia driver or to see if there is an update or patch available for the version of the Arduino IDE that you are using.

Attigliuzzo, 2023-01-19:
@kittaakos: Yes, it runs succesfully with that option or as @Mahiuha wrote, it is usable under Xorg.

closed 2023-01-19 by Attigliuzzo