arduino/arduino-ide #1818: No ports reported on macos ventura 13.1 apple silicon arduino 2.0.3

nickdesaulniers, 2023-01-15:
[via](https://forum.arduino.cc/t/one-mac-sees-my-arduinos-the-other-doesnt/1076201/9?u=ndesaulniers)

I have a board attached but I'm not seeing any ports under Tools. The ports option isn't even in the list.

```sh
$ ls -l /dev/(cu|tty)* | grep modem
crw-rw-rw- 1 root  9,   3 Jan 14 17:29 /dev/cu.usbmodem101
crw-rw-rw- 1 root  9,   2 Jan 14 17:29 /dev/tty.usbmodem101
```
I'm pretty sure that's the device I'm trying to connect to.  I also see it listed under About this Mac > More Info > System Report > USB.

I'm using the board manager URL: https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_dev_index.json from https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/.

nickdesaulniers, 2023-01-15:
Ok, so we were able to resolve the forum thread (at least my issue).  So it looks like if you run a binary compiled for x86_64-apple-darwin on apple silicon which expects aarch64-apple-darwin (err, maybe aarch64e, which is their own ABI for BTI/PAC IIRC), then you will not get a prompt to install rosetta.  You need to manually install rosetta, which can be done from the command line via `softwareupdate --install-rosetta`.

So I guess it's a matter of contacting expressif or whomever to provide cross compilers for aarch64e-apple-darwin hosts in their board manager files.  Not sure how pervasive this is, but perhaps the arduino ide could trigger the rosetta install for the time being?  That would have been nice.

nickdesaulniers, 2023-01-15:
Sorry, it's `arduino-fwuploader` that's first failing (see [the forum post](https://forum.arduino.cc/t/one-mac-sees-my-arduinos-the-other-doesnt/1076201/13?u=ndesaulniers) for the stack trace) which is a node.js app, IIUC.

nickdesaulniers, 2023-01-15:
Not sure where the source for `arduino-ide-extension` are to report the issue to.
https://github.com/arduino/arduino-ide/blob/197cea2a60a9e6eed85be3069ea7a5bd88f963f8/electron-app/package.json#L24
https://www.npmjs.com/search?q=arduino-ide-extension

army11b1980, 2023-01-17:
Very much looking forward to a resolution for this issue.
Apple M1
OS Ventura
IDE 2.0.3 for Mac
Rosetta 2 installed
!!Bad CPU Type!!

army11b1980, 2023-01-18:
@per1234 I have Rosetta 2 installed. I still have this error.

per1234, 2023-01-19:
@army11b1980 please post the full and exact output that contains the "Bad CPU Type" error.

army11b1980, 2023-01-19:
@per1234 
Failed uploading: cannot execute upload tool: fork/exec /Users/michaelsmacstudio/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude: bad CPU type in executable

per1234, 2023-01-23:
> ```text
> /Users/michaelsmacstudio/Library/Arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude: bad CPU type in executable
> ```

OK, this is a completely different problem that happens to produce the same error message. The problem here is you are using a very old version of AVRDUDE from when Arduino only distributed 32-bit macOS builds of that tool. Support for 32-bit applications was dropped in macOS 10.15 Catalina:

https://support.apple.com/en-us/HT208436

So when you attempt to run a 32-bit application you get this "`bad CPU type in executable`" error.

Since your problem is unrelated to the subject of this bug report, and is also not related to the codebase hosted in this repository, it will not be appropriate to continue the discussion here. But if you make a post on the Arduino Forum, I'll be happy to provide you with assistance in solving the problem:

https://forum.arduino.cc/c/using-arduino/installation-troubleshooting/18

per1234, 2023-02-19:
Closing as resolved to the extent that is feasible through the addition of a dedicated article in the Arduino Help Center:

https://support.arduino.cc/hc/en-us/articles/7765785712156-Error-bad-CPU-type-in-executable-on-macOS

I have been closely monitoring GitHub and Arduino Forum for reports and fortunately it seems to not affect many users. So hopefully that will be sufficient. I also hope that Apple might eventually enhance the user experience under these conditions with something similar to the Rosetta installation offer dialog that shows when starting an x86 application.

closed 2023-02-19 by per1234