arduino/arduino-ide #1819: Instal fails.

firepower9966, 2023-01-16:
### Describe the problem

    ID                  Branch     Op     Remote      Download
 1. [✗] cc.arduino.IDE2     stable     i      flathub     1,0 kB / 200,6 MB

Error: While pulling app/cc.arduino.IDE2/x86_64/stable from remote flathub: Invalid checksum for static delta 99e71b2b085273ee56ccbca19e88f72d47851c15a0b0c669614e0813b58b150b
error: Failed to install cc.arduino.IDE2: While pulling app/cc.arduino.IDE2/x86_64/stable from remote flathub: Invalid checksum for static delta 99e71b2b085273ee56ccbca19e88f72d47851c15a0b0c669614e0813b58b150b

### To reproduce

click install.
get 404 error

### Expected behavior

Arduino IDE 2 installs

### Arduino IDE version

stable 

### Operating system

Linux

### Operating system version

21.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-01-16:
Hi @firepower9966. Thanks for taking the time to submit an issue. The ["cc.arduino.IDE2" Flatpak package] is a 3rd party project and Arduino doesn't provide support for it.

Please download the official Arduino IDE from the "Software" page of arduino.cc:

https://www.arduino.cc/en/software

Reports of any problems you find with the official Arduino IDE 2.x package are welcome here.

closed 2023-01-16 by per1234

firepower9966, 2023-01-16:
Thanks for the prompt reply, I installed it using appimage.
I have also emailed the issue to flathub support.

thanks you
Neil