arduino/arduino-ide #1822: xcrun popup on IDE 2.0 macos

albmef, 2023-01-19:
### Describe the problem

Hi 
i get a xcrun not installed popup just after selecting a board. but i won't install the bloated XCODE toolchain never again.

versions:
intel macbook pro
macos ventura v13.1
Version: 2.0.4-nightly-20230119
Date: 2023-01-19T03:05:32.636Z
CLI Version: 0.29.0

best regards,

### To reproduce

step to reproduce
- download the dmg and install arduino IDE for intel mac
- select a board, say arduino UNO (also tested with ESP32 and ESP8266 boards)
- get the popup, clic cancel
- open an example, say blink
- get the popup, clic cancel
- verify/compile sketch
- compilation done no errors

### Expected behavior

get the popup after selecting a board or after opening an example

### Arduino IDE version

2.0.4-nightly-20230119

### Operating system

macOS

### Operating system version

ventura 13.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

Mahiuha, 2023-01-19:
The error message "xcrun not installed" suggests that the Xcode Command Line Tools are not installed on your Macbook Pro. These tools are required for certain command-line tools that are used by the Arduino IDE, such as the avr-gcc compiler.

You can try installing the Xcode Command Line Tools by running the following command in Terminal:
**xcode-select --install**

Alternatively, you can try installing the Command Line Tools for Xcode manually from the Apple Developer website.

It is also possible that the version of the Arduino IDE you are using is looking for a specific version of the Xcode Command Line Tools that is not compatible with the version you have installed. Updating or downgrading the version of the Arduino IDE could help. It is also possible to use alternative toolchains such as avr-gcc, binutils-avr, avr-libc,etc.. which are not dependent on xcode and can be installed via package managers like Homebrew or Macports.

albmef, 2023-01-20:
Dear Joseph
before answering to you i performed some test at my side.
i opened a blinky example and i was able to compile and upload the sketch to both a genuine arduino uno R3 board and a genuine LOLIN D1 mini board (ESP8266), without the need of installing xcode CLI tools on my mac. this confirm that xcode CLI tools are not mandatory for using the IDE with these two targets.
if needed i could test with genuine raspberry pi  PICO boards. let me know if you are interrested in the results

best regards,
Alban

musinou, 2023-01-25:
I gave an Arduino workshop and  3 out of 8 participants had this issue after installing Arduino for the first time. One had still the issue with a 1.8 install after he had installed version 2.0.3. None of the participant had xcode installed.

Mahiuha, 2023-01-25:
> Dear Joseph before answering to you i performed some test at my side. i opened a blinky example and i was able to compile and upload the sketch to both a genuine arduino uno R3 board and a genuine LOLIN D1 mini board (ESP8266), without the need of installing xcode CLI tools on my mac. this confirms that xcode CLI tools are not mandatory for using the IDE with these two targets. if needed i could test with genuine raspberry pi PICO boards. let me know if you are interrested in the results
> 
> best regards, Alban

It sounds like the issue you are experiencing is related to the xcrun command and may be specific to certain boards or configurations. The fact that you were able to successfully compile and upload a sketch to both an Arduino Uno R3 and a LOLIN D1 mini board without installing the Xcode Command Line Tools suggests that the issue may be specific to the board or configuration you are using. If you are interested in testing with genuine Raspberry Pi PICO boards, it may be helpful to share the results of those tests with the community to further investigate the issue.

ishouldmomo, 2023-02-23:
I get this pop-up too (macOS Ventura 13.2.1).
It occurs whenever I open the IDE, and periodically after that. I haven't been able to determine a cause - it doesn't necessarily happen when changing board as described in the original report.
The IDE compiles & uploads (and otherwise works) perfectly as far as I can tell without installation of the tools.

WestfW, 2023-03-04:
I get this as well, running catalina on a MacBook Air, and IDE 2.0.4, with only the AVR and Arduino SAMD boards installed.

It is NOT ACCEPTABLE for the Arduino IDE to require installation of 12G+ of XCode!  (Especially since the main functionality seems to work fine if the request for installation is canceled.)

WestfW, 2023-03-14:
I've been doing some poking, since I have access to several varieties of Mac, with and without XCode installed.
First of all, xcrun is a utility whose purpose is to find XCode-related applications even if they are installed in strange places.
The man page is here: https://www.manpagez.com/man/1/xcrun/

xcrun seems to be quite overzealous; execute a command like "xcrun ls" from a Terminal window, and you'll get an error message that Xcode isn't installed, followed by the same popup window.  :-(

xcrun is a utility in /usr/bin/xcrun

I tried installing a /usr/local/xcrun that was a shell script that logged the arguments, and this stopped the popups from Arduino without seeming to damage anything, but it didn't result in any useful log messages (I'm not surprised.  Presumable xcrun is exec()ed, and can't be a shell script, so it just fails.)  Using an actual  binary with logging capability will take me a while, since I don't actually know how to write Mac applications.  (Maybe plain unix/C will work?)

There are a couple of open source implementations of xcrun.

I don't see any logging of the current errors in any of the usual places, and can't tell if setting xcrun_logging in the environment actually does anything useful (perhaps I'm doing it wrong.)

Hypothesis; something in the  IDE is using xcrun when it should just be using exec.
Interestingly, I can't find any occurrence of xcrun in the source code in either the Arduino code or Thea.  I couldn't figure out how eclipse source code is arranged, so I gave up before that :-(

hollandaisedays, 2023-06-12:
I'm having this same issue on MacBook Pro.  Is there any update or fix to this that has been found yet?  Looks like this chain has been stagnant for awhile, but not closed yet.  Is the only work around to click 'cancel' each time it pops up?  Quite annoying if you are using the application often...

dstulken, 2023-06-30:
Same issue here.  Fresh install of Arduino IDE 2.1.0, on a fairly fresh install of MacOS Ventura 13.4.1 on an Intel Mac mini.

I got the message after installing the ESP32 board set.  After declining the popup, I see this message in the console:

> xcode-select: note: No developer tools were found, requesting install.
> If developer tools are located at a non-default location on disk, use `sudo xcode-select --switch path/to/Xcode.app` to specify the Xcode that you wish to use for command line developer tools, and cancel the installation dialog.
> See `man xcode-select` for more details.
> 
> exit status 1
> 
> Compilation error: exit status 1

And in the case of ESP32 targets, the compilation cannot continue.    The popup itself informs that the "python3" command is what requires the command line developer tools to be installed.

It looks like installing the Xcode command line tools is actually called out at the very bottom of Espressif's Arduino installation instructions:
https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html

sgreensides, 2023-09-09:
Brand new MacBook Pro
tried to use python3 and got very nearly the same result as in a very similar popup and the message:
python3
xcode-select: note: No developer tools were found, requesting install.
If developer tools are located at a non-default location on disk, use `sudo xcode-select --switch path/to/Xcode.app` to specify the Xcode that you wish to use for command line developer tools, and cancel the installation dialog.
See `man xcode-select` for more details.

per1234, 2023-10-06:
I did an investigation of this problem and identified the source of the `xcrun` invocation that produces this dialog when the user doesn't already have the "Command Line Tools for Xcode" installed.

It comes from the [**clangd** C++ language server](https://clangd.llvm.org/) used for the "context-aware" (AKA "IntelliSense") features of Arduino IDE such as [autocomplete](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-autocomplete-feature). **clangd** uses `xcrun` to check whether the [**Clang** compiler](https://clang.llvm.org/index.html) is installed on the macOS user's system:

https://github.com/llvm/llvm-project/blob/llvmorg-14.0.0/clang-tools-extra/clangd/CompileCommands.cpp#L98-L102

```cpp
  // On Mac, `which clang` is /usr/bin/clang. It runs `xcrun clang`, which knows
  // where the real clang is kept. We need to do the same thing,
  // because cc1 (not the driver!) will find libc++ relative to argv[0].
#ifdef __APPLE__
  if (auto MacClang = queryXcrun({"xcrun", "--find", "clang"}))
```

It is the invocation of `xcrun` in general (nothing to do with the presence or absence of **Clang** on the system that triggers the dialog. So the appearance of this dialog is expected and normal for any user of **clangd** who hasn't installed the "Command Line Tools for Xcode" (either by itself or as part of the full [**Xcode**](https://developer.apple.com/xcode/) application).

In Arduino IDE's usage of **clangd**, **Clang** is not necessary, and in fact you will find that the language server-base features in Arduino IDE work fine even if you decline the installation of "the command line developer tools" as offered by the dialog.

Unfortunately it doesn't appear that the LLVM/clangd developers have provided any way to configure **clangd** to skip the invocations of `xcrun`. If we can't find a workaround, the next best thing will be to add an article to the ["**Arduino Help Center**"](https://support.arduino.cc/hc/en-us), providing an explanation for affected users of the dialog.

per1234, 2023-10-06:
> I see this message in the console:
>
>
> And in the case of ESP32 targets, the compilation cannot continue.    The popup itself informs that the "python3" command is what requires the command line [...]

This message and failed compilation has a different cause (even if it will be solved by the installation of the same collection of tools). Typically all dependencies are installed along with an Arduino boards platform installation made via Boards Manager. However, the developers of the "esp32" boards platform took a different approach. They expect macOS and Linux users to have Python 3 pre-installed on their system and depend on that in [compilation commands](https://github.com/espressif/arduino-esp32/blob/release/v2.x/platform.txt#L29) and the upload commands that use [esptool](https://github.com/espressif/arduino-esp32/blob/release/v2.x/platform.txt#L20) or [espota](https://github.com/espressif/arduino-esp32/blob/release/v2.x/platform.txt#L23).

In this case, since it is a dependency of the platform rather than Arduino IDE, the mitigation would need to be made by an entry in the platform's installation documentation, which is off topic for this repository.

hex705, 2024-03-09:
HI, 

The online info on this is contradictory.  Some say you don't need xcrun, others say you do.  

I just installed ide 2.3.2 and esp32 for coding an adafruit huzzah32 feather.
I am using  osx 12.5 on an M1 mini. 

First code upload attempt in ide2.3.2  gave xcrun error (screen shot not text ( sorry solving for students)).  It did not throw a pop-up which i have seen on other platforms / version installs. 

the exit status 1 leaves no choice but to install xcode command line tools.  (at least given my knowledge and what I can see). 

I did this and the IDE now uploads to feathers just fine.

If there is a solution that does not need the xcode install, is it documented officially somewhere?

It seems naively to me that xcode is required as it is the solution (most arduino users are not full software devs -- simple solutions matter).  While xcode tools is large i tell students to just install it because then they can upload code in 2.X on OSX.  

[image]

jrobert-gh, 2024-04-27:
I have the same problem, IDE 2.3.2, macOS Sonoma, iMac/M3.

I had been using versions of IDE 2 since 9 Apr '21 / 2.0.0-beta 4 on an iMac/Intel through 24 Feb '24, and macOS Mojave and Monterey, and hadn't ever seen this popup.

In April, I upgraded to an iMac/M3, macOS Sonoma. I installed IDE 2.3.2 for Apple Silicon on 12 Apr '24. This popup has been appearing whenever I run the IDE, since that time. I have not noticed any deficits in the operation of the IDE, just the annoyance of finding the popup beneath the IDE when I close it, and having to dismiss it.

All of my Arduino software installs were downloaded from the Arduino website (US).

phillmj, 2024-05-19:
This is totally unacceptable. Brand new M3 Mac, installed Arduino IDE 2.x freshly, and now, CONSTANTLY CONSTANTLY get this horrible message demanding I download command line developer tools.

OBVIOUSLY this is a problem. I'm glad it doesn't affect all users, but since this has been going on over a year, I assume it's not that important to the Arduino team.

It is a PROBLEM and makes me wanna rip Arduino out of my system, but I'm an Arduino developer, so I can't.

PLEASE take this seriously and find a way to disable this horrible little troll popup. I am not willing to download 12GB of Xcode because of a glitch in Arduino IDE.