arduino/arduino-ide #1825: Enable rename and delete options for Cloud Sketch and/or any of its files

davegarthsimpson, 2023-01-20:
Right now the UI for Cloud Sketch does not allow the deletion of any of the files open in tabs or the Sketch itself.

requires investigation on allowed API calls and their repercussions

/Users/ubidefeo/Library/Application Support/CleanShot/media/media_NJZguuDWgY/CleanShot 2022-12-23 at 17.18.54@2x.png[image]

kittaakos, 2023-01-20:
> What does the deletion/rename intend to do here? Is it a brute force modification on the remote sketch when the user executes it or is this operation changes the local sketch only then the user has to push the changes to the remote to have the desired delete/rename effect?

> Whenever a new file is created or file content changes, the changes are not automatically applied to the remote content but only to the local cache and users have to push the changes manually. Does IDE2 want to provide different behavior for rename/delete and new/update?

> Can you please help define the expected behavior @91volt? Maybe you had the chance to discuss it with @ubidefeo when you designed the new feature. Thank you!

> [image]

kittaakos, 2023-01-20:
`Rename`/`Delete` must be avail from the sketchbook view.

> [image]

kittaakos, 2023-01-20:
Related bugs:

1. pull deleted sketch:
   - Create cloud sketch,
   - Open IDE2. The new sketch is visible,
   - Delete the cloud sketch from the browser,
   - Pull cloud sketch in IDE2.

```
Uncaught (in promise) Error: Not found: /yyy444.
    at CreateApi.stat (create-api.ts:236)
    at async CreateFsProvider.stat (create-fs-provider.ts:92)
    at async FileService.doResolveFile (file-service.ts:499)
    at async FileService.resolve (file-service.ts:477)
    at async CloudSketchbookTree.resolveFileStat (cloud-sketchbook-tree.ts:427)
    at async CloudSketchbookTree.resolveChildren (file-tree.ts:38)
    at async FileNavigatorFilter.filter (navigator-filter.ts:51)
    at async CloudSketchbookTree.resolveChildren (sketchbook-tree.ts:25)
    at async CloudSketchbookTree.resolveChildren (cloud-sketchbook-tree.ts:382)
    at async CloudSketchbookTree.refresh (tree.ts:306)
```

----

2. Renamed cloud sketch remains in the cache folder
 - Create a cloud sketch,
 - Open IDE2, pull the newly created cloud sketch, and open it in a new window,
 - Rename cloud sketch in the browser,
 - Pull the sketch. Nothing happens, there is an error in the console:
```
uncaught (in promise) Error: Not found: /rrrr_tttt.
    at CreateApi.stat (/Users/a.kitta/dev/g…t_l-bf9e6e.js:11592)
    at async CreateFsProvider.stat (/Users/a.kitta/dev/g…t_l-bf9e6e.js:11893)
    at async FileService.doResolveFile (/Users/a.kitta/dev/g…e-service_js.js:270)
    at async FileService.resolve (/Users/a.kitta/dev/g…e-service_js.js:254)
    at async CloudSketchbookTree.resolveFileStat (/Users/a.kitta/dev/g…t_l-bf9e6e.js:22416)
    at async CloudSketchbookTree.resolveChildren (/Users/a.kitta/dev/g…og_index_js.js:1924)
    at async FileNavigatorFilter.filter (/Users/a.kitta/dev/g…or-widget_js.js:988)
    at async CloudSketchbookTree.resolveChildren (/Users/a.kitta/dev/g…t_l-bf9e6e.js:24272)
    at async CloudSketchbookTree.resolveChildren (/Users/a.kitta/dev/g…t_l-bf9e6e.js:22382)
    at async CloudSketchbookTree.refresh (bundle.js:53231)
```
 - Sync the cloud sketchbook view. A sketch is pulled (the new cloud sketch with its new name), but the currently opened sketch is not visible in the sketchbook tree. After the pull the current cloud sketch is dangling. The sketch with its old name remains in the cache folder under `directories.data`/CloudSketchbook/ArduinoCloud`
 - The IDE2 is fully functional (verify, upload are OK)
 - Quit + start IDE2, the dangling sketch opens, IDE2 is working, but the sketch is not visible the sketchbook view.

kittaakos, 2023-01-20:
Another unhandled use-case for sketch deletion:
 - Open IDE2,
 - Create a new sketch,
 - Click on the `...` on the tab toolbar, select Delete, click OK in the dialog, and see the notification for a subsecond
[image]
 - Nothing happens, no errors.
 - Deletion of temp sketches should be disabled, or the deletion should work.

91volt, 2023-01-20:
>What does the deletion/rename intend to do here? Is it a brute force modification on the remote sketch when the user executes it or is this operation changes the local sketch only then the user has to push the changes to the remote to have the desired delete/rename effect?

We should trigger the API call when the user delete the sketch folder. This is because if we delete the local version only the user wouldn't have any entry point to trigger the "push" of this deletion to Arduino servers. 

I don't see any issue with this behavior, is the same logic we have applied when a user create a new Cloud Sketch from the IDE 2: the API are called directly after the user create the Cloud sketch from IDE 2, he doesn't have to manually push it after creation.

@kittaakos Do you agree?

P.s. I didn't have any chance to discuss it with @ubidefeo yet.

kittaakos, 2023-01-30:
Getting the sketches is not reliable. I use the [`GET /create/v2/sketches`](https://api2.arduino.cc/create/docs#!/sketches95v2/sketches_v2_search) endpoint. The response is:

```json
[
    {
        "board_fqbn": "",
        "board_name": "",
        "board_type": "",
        "created_at": "2023-01-30T14:32:25.218987Z",
        "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/rrrr2",
        "id": "aa6e1ab0-46d1-44ed-86f1-1e2407edb1ad",
        "is_public": false,
        "libraries": [],
        "modified_at": "2023-01-30T14:46:32.634Z",
        "name": "rrrr2",
        "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/rrrr2",
        "tutorials": null,
        "types": null,
        "user_id": "b036b0a3-ef0e-44b1-af37-3c756f20fdf0"
    },
    {
        "board_fqbn": "",
        "board_name": "",
        "board_type": "",
        "created_at": "2023-01-23T16:08:38.100336Z",
        "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/xxx/sketch_jan23bl",
        "id": "8946cc38-e832-4be6-a510-81351cf24506",
        "is_public": false,
        "libraries": [],
        "modified_at": "2023-01-23T16:08:40.547Z",
        "name": "sketch_jan23bl",
        "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/xxx/sketch_jan23bl",
        "tutorials": null,
        "types": null,
        "user_id": "b036b0a3-ef0e-44b1-af37-3c756f20fdf0"
    },
    {
        "board_fqbn": "arduino:avr:ethernet",
        "board_name": "Arduino Ethernet",
        "board_type": "serial",
        "created_at": "2022-10-25T14:02:12.360033Z",
        "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/alma",
        "id": "7258bf5a-e1ad-45ae-82bc-bfde42071cb7",
        "is_public": false,
        "libraries": [],
        "modified_at": "2023-01-16T14:51:12.480999Z",
        "name": "alma",
        "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/alma",
        "secrets": [
            {
                "name": "SECRET_PASSWORD",
                "value": "xxxx"
            }
        ],
        "tutorials": null,
        "types": null,
        "user_id": "b036b0a3-ef0e-44b1-af37-3c756f20fdf0"
    },
    {
        "board_fqbn": "",
        "board_name": "",
        "board_type": "",
        "created_at": "2023-01-16T10:57:24.474541Z",
        "href": "/create/v2/files/d/8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/korte",
        "id": "8d1c721b-dd40-4ad6-8a95-a725032905f9",
        "is_public": false,
        "libraries": [],
        "modified_at": "2023-01-16T13:46:54.140329Z",
        "name": "korte",
        "path": "8a694e4b83878cc53472bd75ee928053:kittaakos/sketches_v2/korte",
        "secrets": [
            {
                "name": "SECRET_PASSWORD",
                "value": "xxx"
            }
        ],
        "tutorials": null,
        "types": null,
        "user_id": "b036b0a3-ef0e-44b1-af37-3c756f20fdf0"
    }
]
```

But I have only three sketches online. Which is the truth; the server response or the online editor?

[image]

kittaakos, 2023-02-01:
[Here](https://github.com/arduino/arduino-ide/pull/1833#issuecomment-1411778181) is another server bug

commit 84e9ea3f (2023-01-31): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
commit b021d3b9 (2023-02-01): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
commit 88043328 (2023-02-01): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
commit 31cc6042 (2023-02-01): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
closed 2023-02-15 by kittaakos