arduino/arduino-ide #1826: Remote > Cloud. Change name and iconography

davegarthsimpson, 2023-01-20:
All the occurrences of the word Remote needs to switch to Cloud.

Top bar menu:
**New Cloud Sketch**
[image]

**Show(Hide Cloud Sketchbook)**
[image]

CTA Sketchbook Sidebar Panel
**NEW CLOUD SKETCH**
[image]

Tooltip
**Cloud Sketchbook**
[image]

Iconography:
[image]

[Link to figma](https://www.figma.com/file/HoD3yMfiiCPBJpxX0U4jAl/IDE-2---Arduino-Cloud-Sketches-update?node-id=6%3A326104)

kittaakos, 2023-01-31:
> ### Additional context
> [Link to figma](https://www.figma.com/file/HoD3yMfiiCPBJpxX0U4jAl/IDE-2---Arduino-Cloud-Sketches-update?node-id=6%3A326104)

From the design:

> To help to recognize if the sketch is Cloud or Local a new Sketch workspace should always open up with the Sketchbook open at the right location (local or cloud) with the current file listed item selected

This will only sometimes work and needs refinement.
 - First off, when creating a new sketch, it's neither in the sketchbook nor in the cloud cache, so IDE2 cannot show anything.
 - Second, IDE2 stores and restores the layout. When I save a sketch and close IDE2, the Theia framework will restore the previous layout on the next app start. @91volt, please confirm if you want the devs to change this behavior and always open the sketchbook view, and select the active tree. Thanks!

kittaakos, 2023-02-09:
Depends on https://github.com/arduino/arduino-ide/issues/1733

kittaakos, 2023-02-10:
Menu item icons are missing: https://github.com/arduino/arduino-ide/pull/1868

commit 84e9ea3f (2023-01-31): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
commit b021d3b9 (2023-02-01): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
commit 88043328 (2023-02-01): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
commit 31cc6042 (2023-02-01): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826