arduino/arduino-ide #1828: Window menu is missing on macOS

rgov, 2023-01-22:
### Describe the problem

In every multi-window macOS app there is a "Window" menu (to the left of "Help") for switching between open windows:

[image]

The Arduino IDE app doesn't have this.

### To reproduce

Compare Arduino IDE to any other multi-window Mac app.

### Expected behavior

There should be a "Window" menu listing the open sketches.

### Arduino IDE version

2.0.3

### Operating system

macOS

### Operating system version

13.1

### Additional context

While the app loads there is actually a "Window" menu amongst the default set of menus, but it gets replaced by the "Tools" menu.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

rgov, 2023-01-22:
There are some [Electron docs on standard menus](https://github.com/electron/electron/blob/main/docs/api/menu.md#standard-menus).

kittaakos, 2023-02-27:
Duplicate of https://github.com/arduino/arduino-ide/issues/1183?