arduino/arduino-ide #1829: switching between ino files or tabs

iddq, 2023-01-23:
### Describe the request

Switching between ino files is sometimes very difficult when there are many ino files in a project because the navigation bar is too thin. Please create a left and right arrow button at both end of the line.

### Describe the current behavior

see above

### Arduino IDE version

2.0.3

### Operating system

Linux

### Operating system version

Debian

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2023-01-23:
Hi @iddq. When you say "navigation bar", do you mean the scrollbar circled here":

[image]

I have also noticed this is far too thin to comfortably use.

I much prefer the size of the scrollbar on VS Code:

[image]

Would you consider a reasonably sized scrollbar like that an acceptable alternative to the arrow buttons you proposed?

---

I also notice the scrollbar has an odd behavior of changing thickness depending on the relative horizontal mouse position:

[image]

It seems like a bug because I can't imagine anyone would intentionally design a UI element to have such a nonsensical behavior.

kittaakos, 2023-01-23:
> Please create a left and right arrow button at both end of the line.

IDE2 should have a wider scrollbar but no additional arrows or actions, as requested.

The code navigation is possible in the following ways:
 - You can scroll over the tabs. You don't need to grab the scroll bar. At least this is working great on macOS.
 - You can also use `Ctrl/Cmd`+`P` to list all files in the current sketch and open one of them,
 - You can select a file from the context menu from the `...` tab-bar toolbar, and
 - You can navigate with `Ctrl/Cmd`+`Alt`+`left/right`. (https://github.com/arduino/arduino-ide/issues/1786)

These seem sufficient, but they need to be documented.

[image]

https://user-images.githubusercontent.com/1405703/214019563-b9337d2e-6774-4ddd-b2ca-ce62c7b2a5e7.mp4

https://user-images.githubusercontent.com/1405703/214019597-5839a4da-bf85-4923-8152-5256ad0848a8.mp4

iddq, 2023-01-23:
Could you please put this CTRL+P search into ... menu or into main menu? It would be useful for new users.