arduino/arduino-ide #1830: find through all files

iddq, 2023-01-23:
### Describe the request

The find function should search all project files, not just the currently focused file.

In addition, it would be nice if the result of search were displayed in a small box with all occurrences in filename:function name format, and were clickable for navigation.

### Describe the current behavior

see above

### Arduino IDE version

2.0.3

### Operating system

Linux

### Operating system version

Debian

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2023-01-23:
Hi @iddq. Thanks for your suggestion.

I'm happy to tell you that this is already possible. Simply click the magnifying glass icon on the "Activity Bar" on the left side of the Arduino IDE window to open the "**Search**" panel. I think you'll find it offers a very capable search and replace functionality.

[image]

closed 2023-01-23 by per1234

iddq, 2023-01-23:
Please create a "Find All" menu item for it in Edit menu, Thank you.