arduino/arduino-ide #1832: Make UI friendlier for larger projects - show only tabs for files that are open, implement some better file navigator on side

benapetr, 2023-01-25:
### Describe the request

Right now IDE is barely useable for very large projects. I have some ESP32 projects that consist of hundreds of .cpp and .h files in complex folder hierarchy. Some files in different folders are in different C++ namespaces and have same file name as other files in different folders.

Such project result in hundreds open tabs. Files with same name are not even shown in drop down menu on side. Switching between files is very complicated as you need to traverse hundreds of tabs on the top and often you only need to work with few files.

Other IDEs like Visual Studio of MPLAB-X support very good file navigator on side where you can display whole hierarchy of project and easily switch between files. Tabs are created only for files that are actually open, so the IDE doesn't need to open every single file for the project.

### Describe the current behavior

Right now IDE is barely useable for very large projects. I have some ESP32 projects that consist of hundreds of .cpp and .h files in complex folder hierarchy. Some files in different folders are in different C++ namespaces and have same file name as other files in different folders.

### Arduino IDE version

2.0.3

### Operating system

macOS

### Operating system version

12.3.1 (21E258)

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

benapetr, 2023-01-25:
awkward workaround is to download Visual Studio Code, move all code to "src/" subfolder which makes ArduinoIDE compile it while hiding the files in itself and write the source code in Visual Studio Code while using Arduino IDE only for compiling and uploading.

kittaakos, 2023-01-25:
There is a "hidden" file explorer in the app.
 - <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>P</kbd>
 - Type `Toggle Explorer View` and press <kbd>Enter</kbd>
 - You won't be able to close the view (https://github.com/arduino/arduino-ide/issues/1220)

https://user-images.githubusercontent.com/1405703/214600652-9356f850-ae53-4ef0-8cf5-45bbfd06ec0f.mp4

If you're working on a larger project, you might want to check out the [`Outline` view](https://github.com/arduino/arduino-ide/issues/1157#issuecomment-1178774800) too. The Arduino language server powers the `Outline` view (see: [`textDocument/documentSymbol`](https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_documentSymbol)), so you must select a board to start the language server, and then the view will be active. You can use <kbd>Ctrl/⌘</kbd>+<kbd>Shift</kbd>+<kbd>O</kbd> to access the document symbols.

https://user-images.githubusercontent.com/1405703/214601651-5387b1f4-ccab-4987-be56-2343c2d08eeb.mp4

benapetr, 2023-01-25:
I am just wondering - is this new Arduino IDE somehow based on Visual Studio Code? This Explorer View looks so much same as VS code explorer, it's using same icons, fonts and design, almost as if it was using same source code?

I also found VSC "arduino" extension, is that somehow related to Arduino IDE?

closed 2023-01-26 by per1234