arduino/arduino-ide #1834: Entire "Tools > Port" menu disabled and no ports discovered

EmmanuelMess, 2023-01-25:
### Describe the problem

No ports discovered:
[image]

### To reproduce

Open Arduino IDE, connect arduino uno board to usb board.

### Expected behavior

It should detect my port and board.

### Arduino IDE version

Version: 2.0.4-nightly-20230125 Date: 2023-01-25T03:03:17.544Z CLI Version: 0.29.0  Copyright © 2023 Arduino SA

### Operating system

Linux

### Operating system version

Ubuntu 20.04

### Additional context

Similar to #605 (closed).

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-01-26:
Hi @EmmanuelMess. Thanks for your report.

In order to gather more information that might help us to troubleshoot your problem, I'm going to ask you to run [the `lsusb` command](https://www.linux.com/training-tutorials/find-usb-device-details-linuxunix-using-lsusb-command/) from the command line and share the output.

1. Open [a terminal](https://ubuntu.com/tutorials/command-line-for-beginners).
1. If it is not already connect the USB cable of your Arduino board to your computer.
1. Type `lsusb`
1. Press <kbd>Enter</kbd>
1. Click and drag the mouse to select all the output in the terminal window.
1. Press **<kbd>Ctrl</kbd>+<kbd>Shift</kbd>+<kbd>C</kbd>**.
   This will [copy](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the selected text to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

per1234, 2023-01-26:
Does your "Uno" have the WCH CH340 USB chip?

[image]

If so, I can see the system recognized it when "connected on the back":

>```text
>Bus 005 Device 012: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter
>```

But not when "connected on the front".

This means the problem with the port is at a system level and not anything related to Arduino IDE. So I'll close the issue here. But you are welcome to request assistance with the system problem over on the Arduino Forum:

https://forum.arduino.cc/

I'll sure we'll be able to help you over there

closed 2023-01-26 by per1234

EmmanuelMess, 2023-01-26:
But connected to the front the arduino ide detects it, but not when connected to the back. So when the system detects it, it doesn't work? That sounds like an ide issue.

EmmanuelMess, 2023-02-19:
Forum link: https://forum.arduino.cc/t/usb-port-not-detected-with-arduino-ide-2/1092204