arduino/arduino-ide #1838: Serial monitor scroll indicator/button

sterretjeToo, 2023-01-28:
### Describe the request

I'm having great difficulty to see if I have selected scrolling in the serial monitor or not. One will usually be able to see it because of the scrolling / non-scrolling of the data but there is the condiction that, when the buffer is full, non-scrolling changes to scrolling without visual indication what was selected.

Please provide a means to see what is active (different icon, different colours, ...). There are three different scenarios that should be indicated
1) scrolling selected
2) non-scrolling selected
3) non-scrolling selected but switched to scrolling

Implementation with different icons is prefered for those that are colour-blind.

### Describe the current behavior

non-scrolling changes to scrolling without indication

### Arduino IDE version

arduino-ide_nightly-20230128_Windows_64bit

### Operating system

Windows

### Operating system version

Win10 Home

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2023-01-28:
Hi @sterretjeToo. Thanks for your suggestion.

>when the buffer is full, non-scrolling changes to scrolling without visual indication what was 
selected.

The suboptimal behavior of Serial Monitor under this condition is tracked at https://github.com/arduino/arduino-ide/issues/1250. I think your suggestion of communicating the reason for this type of scrolling (which is different from autoscroll in that it doesn't scroll to the bottom of the output) is one reasonable option for mitigating that problem, so it would be best as a part of that discussion. Would you mind adding that part of your proposal in the thread of https://github.com/arduino/arduino-ide/issues/1250?

>different colours

Please do this:

1. Select **File > Preferences** from the Arduino IDE menus.
1. Comment here to tell me which theme you have selected from the "**Theme**" menu in the "**Preferences**" dialog.

sterretjeToo, 2023-01-28:
> Please do this:

    Select File > Preferences from the Arduino IDE menus.
    Comment here to tell me which theme you have selected from the "Theme" menu in the "Preferences" dialog.

Theme is Dark (Theia).

I'm aware that it might disappear in the future ;)

per1234, 2023-01-28:
Thanks for the information. There is a previous issue report for that bug: https://github.com/arduino/arduino-ide/issues/1053

The issue was closed because Arduino no longer provides support for the "**Dark (Theia)** and "**Light (Theia)**" themes (https://github.com/arduino/arduino-ide/issues/1283) and the three supported themes do not have the bug.

closed 2023-01-28 by per1234

sterretjeToo, 2023-02-03:
So High Contrast (Theia) will stay? Else I don't know what the 3rd theme is.

[image]

per1234, 2023-02-03:
>High Contrast (Theia) will stay?

Yes. This theme improves the accessibility of Arduino IDE for people with vision impairments so Arduino does support it and there are no plans to remove it.