arduino/arduino-ide #1848: IDE restarts during a start

FernandoGarcia, 2023-01-31:
### Describe the problem

Hi!

I have  noticed the behaviour shown on video all time in IDE 2.0.3.

https://user-images.githubusercontent.com/2778501/215854286-77a9a949-13b3-4c5f-9165-365c0fc4b29a.mp4

Actually I'm using flatpak but I saw this in AppImage too.

I have the full log from terminal attached.

[full.log](https://github.com/arduino/arduino-ide/files/10549784/full.log)

Best regards.

### To reproduce

Open the IDE.

### Expected behavior

Don't restart.

### Arduino IDE version

2.0.3

### Operating system

Linux

### Operating system version

Elementary OS 5

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-02-03:
The IDE2 reloads if you open a sketch that does not exist or is invalid. What was the sketch you wanted to open? Please share the sketch folder path.

> Actually I'm using flatpak

I see an error in the filesystem watcher from the attached log that is related to the flatpack: https://github.com/refi64/zypak.

```
2023-01-24T17:05:58.740Z root ERROR [nsfw-watcher: 139] [139 zypak-helper] Using spawn strategy test 1 as set by environment
```

I do not remember supporting flatpack. If something is wrong with the filesystem watcher, many things can go wrong in IDE2. I suggest using AppImage if you can.

I noticed another issue with the CLI's discovery:

```
INFO[0000] Loaded tool                                   tool="builtin:mdns-discovery@1.0.5"
INFO[0000] Loaded tool                                   tool="builtin:mdns-discovery@1.0.6"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.2.1"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.3.2"
INFO[0000] Loaded tool                                   tool="builtin:serial-discovery@1.3.3"

[...]

2023-01-24T17:05:57.082Z daemon INFO INFO[0002] from discovery builtin:serial-discovery received message type: hello, message: OK, protocol version: 1 
INFO[0002] sending command START_SYNC to discovery builtin:serial-discovery 
INFO[0002] from discovery builtin:serial-discovery received message type: start_sync, message: Cannot START_SYNC: address already in use 
ERRO[0002] Discovery builtin:serial-discovery failed to run: starting discovery builtin:serial-discovery: command failed: Cannot START_SYNC: address already in use 
```

FernandoGarcia, 2023-02-03:
Hi!

I was default sketch, with date in the name as you can see in this image:

[image]

The patch to sketch folder is: `/home/fernandogarcia/Arduino`

The source of the flat pack is:

https://github.com/flathub/cc.arduino.IDE2

FernandoGarcia, 2023-02-28:
I can't see the restart in IDE 2.0.4 but I can see the the double tab yet.

closed 2023-02-28 by FernandoGarcia

joshuag1000, 2023-03-01:
Hi, @kittaakos (apologies for the ping in an issue here I wasn't sure how best to reach you) 

Thanks for triaging flatpak issues. I try to regularly check here to ensure that the flatpak version that i maintain isn't causing any issues for you guys here.
Recently i have really busy and with no way to setup an automatic email filter for issues i havent been able to check.
To reduce any further workload on you guys I'm going to change the flathub issue url to the flathub issues at https://github.com/flathub/cc.arduino.IDE2/issues to ensure any future issues are dealt by me before passing them to your team if necessary. I'm also going to make an unofficial wrapper notice on the flathub page for the IDE mentioning that it is an unofficial wrapper so people are aware.
In the event an issue slips through would it be possible if you could ping/tag me @joshuag1000 before you close the issue so github will send me an email and i can work on fixing the issue that would be amazing!

Apologies for the inconvenience caused.
- J

kittaakos, 2023-03-01:
Hi, @joshuag1000. Thanks for the ping and your dedication to the Flatpack version of IDE2 💪 

> In the event an issue slips through would it be possible if you could ping/tag me @joshuag1000 before you close the issue so github will send me an email and i can work on fixing the issue that would be amazing!

I am happy to adjust how I deal with the issues if it helps the Arduino community using the flatpack build, but I am not 100% sure what you want me to do. I did not close this task. Feel free to contact me via my business email address; it is attached to every commit I made in this repository.

joshuag1000, 2023-03-01:
> Hi, @joshuag1000. Thanks for the ping and your dedication to the Flatpack version of IDE2 muscle
> 
> > In the event an issue slips through would it be possible if you could ping/tag me @joshuag1000 before you close the issue so github will send me an email and i can work on fixing the issue that would be amazing!
> 
> I am happy to adjust how I deal with the issues if it helps the Arduino community using the flatpack build, but I am not 100% sure what you want me to do. I did not close this task. Feel free to contact me via my business email address; it is attached to every commit I made in this repository.

So for you if you were to find an issue that isn't an issue with the IDE itself but you may think could be related to the flatpak build (like #1466 or #1819):
- Send a comment that pings me.  (@joshuag1000)
- And then process the issue like you normally would which in these cases would probably be to close them.

What i will do is then create an issue at https://github.com/flathub/cc.arduino.IDE2/issues where i will fix the issue and push an update.

I've updated the issue links on the flathub website (see: https://beta.flathub.org/apps/cc.arduino.IDE2) so that most issues from the flathub build will be posted to the flathub issue tracker.
This means that i will get the flathub issues without them being posted here and in the event i find any IDE issues within them ill create an issue here so you guys can fix the IDE.
It just means that I am able to provide better support for the flathub package providing better support for those who use ide.

I hope this makes sense.
If you need me to clarify it more we can move this to emails.

kittaakos, 2023-03-01:
> I hope this makes sense.

Sure. Thanks for taking the time and explaining it to me.