arduino/arduino-ide #1853: Autoscroll button in serial monitor should have text to indicate wtf it is.

iplayfast, 2023-02-02:
### Describe the request

I ended up doing an internet search to try to find it. I hate useless icons. How hard it is to put text into the button so it becomes an obvious thing. Two chervons don't mean anything to me, and even though I tried it, I didn't understand that it was toggling autoscroll off and on. 

What I would prefer to see is text that says, "Turn autoscroll on" if autoscoll is off, and "Turn autoscroll off" if it's on. There is room on the bar, so let's use it.

A less useful text would be "Autoscroll" so people would know what the button is for. 

### Describe the current behavior

Current behaviour is mostly hidden.

### Arduino IDE version

2.03

### Operating system

Linux

### Operating system version

Mint 23

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

chrisjj, 2023-02-15:
Hear hear.

nmzaheer, 2023-10-21:
@per1234 Is there a resolution for this? How can I help?