arduino/arduino-ide #1854: Can't open *.c or *.h or any other files

iplayfast, 2023-02-02:
### Describe the problem

I was exploring the grbl source code, [https://github.com/grbl/grbl](url) and found that I could open the grblUpload.ino file, I could not open any of the supporting files, which are all c or h.

I tried using file/open but it would only see ino. I tried drag/drop but I didn't work. 

I was able top using the goto definition right click on an include file, but that trick only worked for includes.

### To reproduce

Try to open a file without the extension ino

### Expected behavior

As the ide editor seems to understand C it should be able to load C.

### Arduino IDE version

2.03

### Operating system

Linux

### Operating system version

mint 23

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-02-02:
Hi @iplayfast. Arduino IDE is a tool for developing Arduino sketches. It can only be used for opening Arduino sketch projects. It is not a general purpose editor, so you can not directly open arbitrary files. This is the intentional design of the application.

I will recommend [**VS Code**](https://code.visualstudio.com/) as a general purpose development editor. The [**Eclipse Theia** IDE framework](https://theia-ide.org/) Arduino IDE 2.x is built on uses components and design philosophy from VS Code, so you will find there is quite a bit of consistency in the user interface between Arduino IDE 2.x and VS Code.

closed 2023-02-02 by per1234

ubidefeo, 2023-02-02:
@iplayfast 
GRBL is not provided as an Arduino project, and is not meant to be edited like one.
It is very different from something like Marlin, where you can edit the configuration of your machine and setup.

GRBL is distributed as an Arduino Library via a downloadable zip file,
and the only thing you can do is open its upload example and upload it to a compatible hardware (usually ATMega328 or ATMEGA1280/2560 and a few more)

If you want instructions here's a validated process
https://www.youtube.com/watch?v=m998bYioHqs

iplayfast, 2023-02-05:
@per1234 You are right, in that the IDE is not a general purpose editor. (that was my complaint!) However, the ino file is just a C++ file, with a weird extension and the Arduino headers build in. It seems to me that using a c++ file should be withing the possibility of this ide. Not to mention library development.

@ubidefeo Grbl was intended only as an example of arduino code that can't be used with the new ide. I can give lots of examples of arduino libraries that can't be edited either!