arduino/arduino-ide #1857: Syntax Highlighting in Editor Stops working partway through file

BrianAtDocumentedDesigns, 2023-02-02:
### Describe the problem

I made a change to my file and syntax highlighting stopped. I tried closing the IDE, renaming the file, even installing and using a fresh copy of the nightly and nothing seems to work. It looks like the editor treats everything after a particular line (Line 268) as a comment.

The file compiles with no problem, however.
[Tacometer_V0.zip](https://github.com/arduino/arduino-ide/files/10570344/Tacometer_V0.zip)

### To reproduce

I can't reproduce it - it just happens to the file and stays that way

### Expected behavior

I would expect syntax highlighting to not stop working

### Arduino IDE version

IDE 2.03 or Feb 2 2023 nightly

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

BrianAtDocumentedDesigns, 2023-02-02:
Just discovered that if I comment out the whole switch case statement, syntax highlight reappears after

kittaakos, 2023-02-03:
Hi, thanks for reporting it and attaching the source; I could reproduce the defect. I looked into this mysterious issue, and I can confirm the syntax highlight of your code is also broken with the [built-in extension from VS Code](https://open-vsx.org/extension/vscode/cpp). This is what IDE2 uses under the hood:

https://github.com/arduino/arduino-ide/blob/f5621db85d2d9186c45f7580636f590595663c1d/package.json#L78

The highlighting of the sketch from VS Code with the built-in C++ extension is also broken:

[image]

But I have good news: formatting your code in IDE2 fixes the syntax highlighting. It works if you insert a space after the `case` keyword. Or you can format the entire document, and the syntax coloring will be correct in the sketch. I hope this helps for the time being.

https://user-images.githubusercontent.com/1405703/216551996-8da87696-b6e4-4d9c-a24f-41271e7e4480.mp4

So that you know, we are aware of a few issues regarding the syntax coloring: https://github.com/arduino/arduino-ide/issues/380.

closed 2023-02-03 by per1234

BrianAtDocumentedDesigns, 2023-02-03:
Thanks for the reply. I don't know why this is flagged as "off topic" though. As a general rule, software which doesn't work as expected should be reported. 

It is clearly a bug. Telling somebody a bug report is "off topic" discourages bug reports.

per1234, 2023-02-03:
@BrianAtDocumentedDesigns it is marked "off topic" because it is not our bug. Arduino IDE 2.x is built on many other software projects. You may experience faults in Arduino IDE that are caused by a bug in one of those other software projects. In that case, the bug must be reported to the issue tracker of the codebase that contains the bug.

Of course, we don't expect the casual users will always be able to differentiate a fault originating from a bug in a dependency from a fault originating from a bug in Arduino's own codebase, so it is completely fine to report faults you experience when using Arduino IDE here, but if we determine the bug is not in our codebase, then we will close the issue because it is off topic here.

In extreme cases where a bug in a dependency is seriously impacting the usability of Arduino IDE and the upstream fix might take some time, we might try to make a patch here in this codebase. But this is not such a case. The cosmetic defect is specific to your non-standard code formatting (I was quite surprised this is even valid code), so this is not considered a critical defect that requires action on our part.

You are welcome to report the bug to the upstream project:

https://github.com/jeff-hykin/better-cpp-syntax

Please make sure to do the following first:

- Verify the fault still occurs when using the development version of that project
- Verify that there isn't an existing report in their issue tracker

Make sure to include the following in your report:

- A [minimal, reproducible, verifiable example (MCVE)](https://stackoverflow.com/help/minimal-reproducible-example) of the fault (the sketch you shared here contains a huge amount of irrelevant code)
- A reference to the section of the C++ specification that supports this odd omission of a space after the `case` statement as specification compliant

JustTryingToGetSomeWorkDone, 2023-11-30:
I think that if it's marked off topic the person marking it off topic should be responsible for, _at the very least_, reporting it to the upstream project. Users who report bugs here clearly do not know the bug belongs to another project. However the people choosing to use other projects in their own projects (VSCode code in Arduino IDE for example), do know and should feel responsible for reporting bugs in the projects whose source they're using that become apparent to them through bug reporting in their own projects. You have recognized this bug belongs to VSCode, you should report it. If we stop reporting bugs because it's annoying, open source software falls apart. Do you want people to stop reporting bugs to you because it's annoying for them to go through the effort? Imagine how few people would use Arduino if it was a HUGE buggy mess that no one ever reported bugs on. No one would use it and the project would die. Report your bugs Per1234.