arduino/arduino-ide #1860: IDE Re-prompts for admin credentials for each user

9072997, 2023-02-03:
### Describe the problem

I have a system install (using the MSI with ALLUSERS=1) of the Arduino IDE on a lab of computers. When you launch the IDE for the first time you get a prompt for admin credentials for "dpinst-amd64.exe". I assume this is to install drivers. It would have been really nice if those got installed as part of the IDE installer, but whatever. I can go around on all the computers and launch the IDE once to let it go ahead and install the drivers.

The problem is the IDE tries to install these drivers on the first launch ***for each user***. So if Alice logs in to Computer 07, it will install drivers the first time she launches the IDE. Then when Bob logs in to Computer 07 it will install them again. Then when Alice logs into Computer 13, it will install them again. As you can imagine, this winds up prompting for admin credentials *a lot* more than it needs to.

It would be great if either
- drivers were installed as part of the installer (which is already elevated)
- The IDE could check if the drivers were installed before trying to install them

### To reproduce

1. Install the IDE for all users (`msiexec /i arduino-ide_2.0.3_Windows_64bit.msi ALLUSERS=1`)
2. Launch the IDE
3. Enter a username & password (or just click "yes" depending on how you have UAC configured) to let it install drivers
4. Sign out of Windows
5. Sign in as another user
6. Launch the IDE
7. Observe that you get another UAC prompt

### Expected behavior

Ideally, drivers are installed as part of the MSI and there are never any UAC prompts

Also acceptable: You get a prompt at step 3 (the first user) but not at step 7 (or for any subsequent users)

### Arduino IDE version

2.0.3 (msi) / 2.0.4-nightly-20230203 (zip)

### Operating system

Windows

### Operating system version

Windows 10 - 21H2

### Additional context

AFAIK there are not MSI builds for nightly, so in order to test with nightly I just unpacked the ZIP to `C:\Program Files\Arduino`

Image of prompt:
[image]

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

Elhue, 2023-02-10:
Or give us a zip of the drivers that could be installed in a script as part of the deployment

9072997, 2023-02-13:
I don't think that would solve the issue, since the IDE prompts to install drivers whether or not they are already installed.

SRQmike, 2023-02-22:
yeah this new 2.x.x  "Board Manager" really messes us up in K-12 environments I am rolling my labs back to 1.8.19 right now to get them working again. if we can get an EDU version with all the basic boards installed that would be huge. Something we can Deploy that works like the 1.8.x versions.

bwalz87, 2023-02-22:
> yeah this new 2.x.x "Board Manager" really messes us up in K-12 environments I am rolling my labs back to 1.8.19 right now to get them working again. if we can get an EDU version with all the basic boards installed that would be huge. Something we can Deploy that works like the 1.8.x versions.

I can agree. 1.8.19 was great. We've reinstalled it because version 2.x.x is terrible for deployments. We DO NOT give out admin rights to anyone but IT staff.

CheezWizTN, 2023-02-27:
I think someone over on this thread figured out how to deploy this: 
https://forum.arduino.cc/t/arduino-ide-2-0-rc-silent-install/960245/1

Lukas-UAUX, 2023-03-20:
That is not really a good solution... We also would really appreciate a deployable version. Having an MSI is already a good step (allthoug it weirdly fails when using Local System as most deployment solutions do...) but it should have a INSTALLEVEL option to just install everything for every user (classrooms are most likely multi user devices and even if not the admin user who installs is most definitely not the same the end user will use).

Lukas-UAUX, 2023-03-20:
... installing stuff to the AppData of a single user at install time (and not at first startup) is always a bad idea as you can never know which user will use the software and if he has the needed privileges.

mmseng, 2023-09-01:
Won't _someone_ think of the (school)children!

mseymour-plym, 2024-09-10:
This is still an issue in September 2024. Having to stick with 1.8.9 for our courses using Arduino IDE.