arduino/arduino-ide #1867: ERROR:broker_win.cc(56)] Error reading broker pipe: The pipe has been ended. (0x6D)

klodoma, 2023-02-07:
### Describe the problem

I just installed the latest version(2.0.3) and it ran the first time. 

When I try to start it now, I get this error:
```
[20560:0207/222807.596:ERROR:broker_win.cc(56)] Error reading broker pipe: The pipe has been ended. (0x6D)
```
[image]

### To reproduce

Install version 2.0.3. Run multiple times

### Expected behavior

The ide does run

### Arduino IDE version

2.0.3

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-02-09:
Hi @klodoma. Thanks for your report.

Unfortunately I don't have a good idea about the cause of this error. I haven't seen this reported before from Arduino IDE users, but I did find some mentions of the same error occurring for the users of other applications.

One thing mentioned was that it could be caused by security software interfering with the application's functioning. Is there anything notable about your antivirus software, firewall, ["localhost"](https://en.wikipedia.org/wiki/Localhost) configuration, or network (e.g., use of a VPN or proxy)?

klodoma, 2023-02-09:
I re-installed it a few times, tried the zip version etc.. 
The very first time it ran cause I got all the request from the firewall etc... 

I don't have any special config regarding firewall, AV. I disabled them, same issue. 

I also found a log file in `c:\Users\andy\AppData\Roaming\Arduino IDE\2023-02-07_log.log` which confirms the run. 
Sometimes when I try to run it, a new empty log file is created in this folder. 

I haven't checked the sources, but does it look up for some configuration files or smth? 

I have a normal PC, I use it for development so it's a tweaked configuration... still... nothing special :)

per1234, 2023-02-09:
OK, thanks for the information. Another thing I saw mentioned is that it could be related to the computer's GPU. I will ask you to try an experiment that might give us an idea of whether that is the case for you.

Start Arduino IDE from PowerShell using this invocation:

```text
& '.\Arduino IDE.exe' --disable-gpu
```

Please comment here to let me know whether the same "`Error reading broker pipe ...`" failure still occurs after you start Arduino IDE using that `--disable-gpu` flag.

>does it look up for some configuration files or smth?

Yes, they are located in these folders:

```text
c:\Users\andy\.arduinoIDE\
```

```text
c:\Users\andy\AppData\Roaming\arduino-ide\
```

It also accesses this folder:

```text
c:\Users\andy\AppData\Local\Arduino15\
```

And the sketchbook folder which usually defaults to this location:

```text
c:\Users\andy\Documents\Arduino\
```

(but maybe slightly different if you have your Documents on Windows OneDrive)

klodoma, 2023-02-09:
Thanks for the detailed answer. 

**Deleting all these folders seems to have fixed the issue.** 

Actually I just realized that even when the `Error reading broker pipe: The pipe has been ended. (0x6D)` was throwed the application started (process was running) it was just probably hanging around. 

I have a backup of these folder, should we dig further or good enough for the moment? 

For me it's ok to close.

closed 2023-09-28 by per1234