arduino/arduino-ide #1874: Emoji emoji not displaying properly

lewixlabs, 2023-02-10:
### Describe the problem

Hi,
my Issue is: emoji not displaying properly on Arduino IDE 2 native, but on cloud editor they are ok.
Maybe a unicode parsing problem?

I attach screenshots
[image]
[image]

### To reproduce

1. Insert any emojy on your code in cloud editor
2. from you arduino ide 2.0 native make a pull of that code (from cloud -> to native)
3. you can notice emoji not displaying properly 

### Expected behavior

Emoji must be correctly displayed on arduino ide

### Arduino IDE version

2.0.4-nightly-20230210

### Operating system

macOS

### Operating system version

Ventura 13.2

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-02-10:
Hi @lewixlabs. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/449.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2023-02-10 by per1234

lewixlabs, 2023-02-10:
Hi @per1234 ok thank you, I didn't find this report you mentioned when I searched on opens issues.