arduino/arduino-ide #1875: Git integration

lewixlabs, 2023-02-10:
### Describe the request

I think a nice and very usefull feature could be a git integration in Arduino IDE 2.x.
Nowdays many IoT/Embedded Arduino project became big, as you know, so I think git integration is necessary.

Feature to add could be (in order):

- [ ] Arduino IDE 2.x git integration for offline projects
- [ ] Arduino IDE 2.x git integration for Arduinno IoT Cloud project: behavior could be
- I pull a cloud a project from cloud web editor
- Arduino IDE 2.x create a local folder to save and synchronize all modules
- Arduino IDE 2.x enable git initialization on this folder

P.S.: I know (and I use) git CLI from shell, but this feature could accelerate development.
Thanks.

Lewix

### Describe the current behavior

Actually, even on nightly build

### Arduino IDE version

2.0.4-nightly-20230210

### Operating system

macOS

### Operating system version

Ventura 13.2

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2023-02-10:
Hi @lewixlabs. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1880.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2023-02-10 by per1234

lewixlabs, 2023-02-10:
Ok thank you @per1234 !
Just a note: ok is not planned, but one day please add complete git support, not "commit only" feature (I read this on #1880 issue 😅)