arduino/arduino-ide #1877: Move Cloud "Login entry point" to the sidebar of IDE2

davegarthsimpson, 2023-02-10:
### Describe the request

We hope to introduce and promote Arduino Cloud to as many IDE2 users as possible. We would therefore like to make the integrated cloud login in IDE2 more visible.

#### Outcome

A user avatar together with a popover menu should be added to the "first level of the interface", in the sidebar. The popover menu should contain Arduino cloud related actions, namely Sign in/out and links to the cloud platform.

The existing avatar should be removed, and the "Cloud account state element" (at the bottom of the cloud sketchbook list) should be augmented as per design spec. 

### Additional context
Figma links here

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2023-02-10:
The new requested _login entry point_  occupies the space from the default [_Account management_](https://code.visualstudio.com/updates/v1_44#_account-management) in Theia/VS Code.

From VS Code:

[image]

It will likely break the IDE2 UI if any future VS Code extension contributing to [`authentication`](https://code.visualstudio.com/api/references/vscode-api#authentication) API is added to IDE2. See the Theia logic [here](https://github.com/eclipse-theia/theia/blob/371ab02dda5c6df2c1381ab68b2f5ae341f7d258/packages/core/src/browser/common-frontend-contribution.ts#L456-L458). Currently, no extensions contribute to this API, so there is no UI issue now.

kittaakos, 2023-02-10:
When `Advanced` > `Show/Hide Remote Sketchbook` is off (`"arduino.cloud.enabled": false`), then IDE2 does not show `File` > `New Remote Sketch`, and the _Remote Sketchbook_ is also hidden from the _Sketchbook_ view.

[image]
[image]

@91volt, does IDE2 want to hide the _login entry point_ from the sidebar? Thank you!

davegarthsimpson, 2023-02-10:
> The new requested _login entry point_ occupies the space from the default [_Account management_](https://code.visualstudio.com/updates/v1_44#_account-management) in Theia/VS Code.
>
> From VS Code: [...]

good point, @91volt if in the future we add something that contributes to the `authentication` api we may need to consider moving the cloud avatar to avoid obscuring the default Theia/VS Code element.

commit 5f8d8644 (2023-02-13): feat: moved login entry point to the side-bar

Closes #1877
commit 0ae5ec37 (2023-02-15): feat: moved login entry point to the side-bar

Closes #1877
PR #1881 (2023-02-20): feat: moved login entry point to the side-bar
### Motivation

Moved the Arduino Cloud login from the tree to the sidebar.

### Change description

New login location:

https://user-images.githubusercontent.com/1405703/218430756-cf160982-f5e3-489d-b90c-890c5e0d9237.mp4

### Other information

Closes #1877

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / updated (for [... cut]
commit 4deaf4fb (2023-02-20): feat: moved login entry point to the side-bar

Closes #1877