arduino/arduino-ide #1882: Sketch deleted when location inside sketch folder selected as rename target

per1234, 2023-02-11:
### Describe the problem

The Arduino IDE editor toolbar contains a "context" menu accessed via the **●●●** icon. This menu includes a "**Rename**" option. The behavior of the "**Rename**" option depends on which type of sketch file tab is selected in the editor:

- If it is a secondary file, the option renames the file alone
- If it is the primary `.ino` file (the leftmost tab), the file and sketch folder are renamed

🐛 If the user selects a location inside the current sketch folder when renaming the primary tab, the sketch may be deleted.

### To reproduce

1. Select **File > New Sketch** from the Arduino IDE menus.
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch with the name `Foo`
1. Click on the "**Foo.ino**" tab in the Arduino IDE editor.
   **ⓘ** This step is required due to https://github.com/arduino/arduino-ide/issues/1847
1. Click the **●●●** icon on the right side of the editor toolbar.
   The editor context menu will open.
1. Select "**Rename**" from the menu.
   The "**Save sketch folder as...**" dialog will open.
1. Select the `Foo` folder from the navigation panel of the dialog.
1. Save the sketch as `Bar`

🐛 Intermittently, the above procedure will cause the the `Bar` sketch to be deleted along with the `Foo` parent folder.

The other times it will not be deleted, but the logs show that the IDE still attempted the deletion:

```text
2023-02-11T22:29:01.644Z sketches-service ERROR Failed to delete sketch at c:\Users\per\Documents\Arduino\Foo.
2023-02-11T22:29:01.645Z root ERROR Request deleteSketch failed with error: EBUSY: resource busy or locked, rmdir 'c:\Users\per\Documents\Arduino\Foo\Bar'
2023-02-11T22:29:01.694Z root ERROR Error occurred when executing the startup task 'arduino-delete-sketch' with args: '["file:///c%3A/Users/per/Documents/Arduino/Foo"].
```

The intermittent outcome seems to be dependent on the timing of whether the deletion of the `Foo` folder is completed before the IDE opens the `Foo/Bar/Bar.ino` file. If the former, the sketch is deleted. If the latter, the deletion process fails due to the IDE creating a lock on it by having the sketch open.

### Expected behavior

There is no valid use case for this procedure. The user will only attempt the above procedure based on a misunderstanding of the nature on an Arduino sketch and how the IDE's "**Rename**" feature works when done on the primary sketch file. These users expect the outcome to be that the primary sketch file alone is renamed, resulting in this structure:

```text
Foo/
└── Bar.ino
```

However, the IDE's "**Rename**" feature intentionally does not work this way due to [the requirement](https://arduino.github.io/arduino-cli/dev/sketch-specification/#primary-sketch-file) that the primary `.ino` file match the sketch folder name.

So the ideal behavior would be for Arduino IDE to:

- Prevent the sketch from being be renamed to a location inside its previous sketch folder.
- Clearly communicate to the user that this is not allowed.

---

If the above is not possible, the second best behavior would be to end with a structure like this:

```text
Foo/
└── Bar/
    └── Bar.ino
```

This will not be the user's expected outcome, but the sketch is not deleted and remains valid.

### Arduino IDE version

79b6b7e

### Operating system

Windows

### Operating system version

11

### Additional context

I am also able to reproduce the fault using the tester build from #1833 (https://github.com/arduino/arduino-ide/pull/1833/commits/b9e9ae9c1353b7e7a615e59377a9dff16ce6c7f0). I found the fault was no longer intermittent, but sometimes has a different characteristic in that the sketch is deleted after existing the IDE. It seems the deletion is postponed when the IDE is not able to complete the process due to a lock:

```text
>>> Finishing scheduled sketch deletions before app quit...
>>> Running sketch deletion c:\Users\per\Documents\Arduino\Foo before app quit...
<<< Deleted sketch c:\Users\per\Documents\Arduino\Foo
<<< Successfully finishing scheduled sketch deletions.
```

---

Originally reported at https://forum.arduino.cc/t/arduino-ide-deleted-all-project-files-after-changing-from-ino-to-cpp/1088672

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

DerangedMind, 2023-02-12:
@per1234 thanks for creating this issue on my behalf! I read through your analysis and I'm concerned that the proposed solution is a bandage on the underlying problem. 

The way I'm seeing the problem, if the main `.ino` file is not found, it will trigger a sketch project initialization which results in the entire file path being reinitialized. 

After looking through some of the source code, I found this in the sketch init service:

https://github.com/arduino/arduino-ide/blob/658f117e93ed326704d95e785279e5a04f559675/arduino-ide-extension/src/node/sketches-service-impl.ts#L89

I believe a solution that would resolve the punishment for this action would be easier to implement. In addition, a function named "getSketches" implies no side-effects, but it's clearly doing some heavy lifting here. Ideally, this should simply do a check and pass the responsibility of resolution to the caller. 

I can take a look at this later if this solution sounds promising. I've got many years of experience working with Theia, so I'm happy to make a contribution, just let me know. Thanks again!

kittaakos, 2023-02-13:
> So the ideal behavior would be for Arduino IDE to:
> 
> * Prevent the sketch from being be renamed to a location inside its previous sketch folder.
> * Clearly communicate to the user that this is not allowed.

IDE2 can do the same as IDE 1.x:

[image]

kittaakos, 2023-02-13:
> 8\. Save the sketch as `Bar`
> 
> 🐛 Intermittently, the above procedure will cause the the `Bar` sketch to be deleted

IDE2 must disallow nested sketch destination location if the source sketch should be deleted, but the real problem is here: https://github.com/arduino/arduino-ide/blob/79b6b7ecc092292ae912eb7fec6bc7ed4f8cea04/arduino-ide-extension/src/browser/theia/workspace/workspace-commands.ts#L150

When renaming an ordinary, non-temp sketch, IDE2 must not wipe the source sketch as the operation is save-as.

per1234, 2023-02-13:
>IDE2 can do the same as IDE 1.x:

Even though I appreciate the sentiment behind the dialog title, I don't think it is appropriate in this context. The user who triggers this dialog is already confused about the nature of Arduino sketches and is unlikely to be in the mindset to appreciate an obscure "Borges" reference. They may try to interpret it from a technical (vs. humorous) perspective and become even more confused.

I would also drop the "this would go on forever" part. This is referencing a potential (it doesn't exist in Arduino IDE 2.x) defect in the application which would cause it to create an infinitely recursive subfolder structure until something breaks in the file/operating system. So this implies that the restriction is only a workaround for a defect instead of an intentional design.

But as far as the idea of using a dialog to communicate the situation to the user, I'm fine with that. 

>When renaming an ordinary, non-temp sketch, IDE2 must not wipe the source sketch as the operation is save-as.

I'm not sure about this. A rename operation is equivalent to a move, where the source is removed. That is the current behavior of the IDE's sketch rename feature. You are proposing changing it to a copy operation.

kittaakos, 2023-02-13:
> Even though I appreciate the sentiment behind the dialog title, I don't think it is appropriate in this context.

We're on the same page 👍 

> I would also drop the "this would go on forever" part.

I am open to any suggestions.

> I'm not sure about this.

It's clearly a bug with IDE2. Check with IDE 1.x:

https://user-images.githubusercontent.com/1405703/218413307-1ae847b3-3334-4637-ad42-ddfc9c318403.mp4

 As a user of any software application, for me, save as does not mean moving and losing the original content but if you think IDE2 should behave differently, I can keep the current logic and let IDE2 wipe the original sketch content after save as.

kittaakos, 2023-02-13:
You're the best. Thank you for the correction! It's not a bug.

commit bec6e927 (2023-02-13): fix: disallow nested sketch on save as

Do not wipe source sketch after save as if not a temp

Closes #1882
commit adf3dcfb (2023-02-13): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #649
Closes #1847
Closes #1882
PR #1833 (2023-02-15): feat: rename, deletion, and validation support
### Motivation

 - Support for sketch folder and code file name validation. Cached cloud sketches use the Create Editor's validation. The rest uses the [spec](https://arduino.github.io/arduino-cli/dev/sketch-specification/#sketch-root-folder).
 - Support for renaming and deleting local and cloud sketches.
 - Renamed `remote` to `cloud` sketch in the UI.

### Change description

### Other information

Closes #1599
Closes #1825
Closes #649
Closes #1847
Closes #1882
Ref #1826 (partially fixes)

### Reviewer checklist

* [ ] PR addresses a single conc [... cut]
commit d68bc4ab (2023-02-15): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #649
Closes #1847
Closes #1882
closed 2023-02-15 by kittaakos