arduino/arduino-ide #1888: Why is there no type: bug label?

chrisjj, 2023-02-15:
### Describe the request

... particularly given the concept of bug is recognised by the Bug Report form. https://github.com/arduino/arduino-ide/issues/new?assignees=&labels=type%3A+imperfection&template=bug-report.yml

That form applies only "type: imperfection", where the word imperfection falls way short of, and is much broader than, bug.

### Describe the current behavior

-

### Arduino IDE version

None

### Operating system

N/A

### Operating system version

None

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2023-02-15:
>the word imperfection falls way short of, and is much broader than, bug.

You just answered your own question. The "**type: imperfection**" label is intended to be applicable to any component of the project, at any severity on the spectrum from "missing comma in the readme" to "[Skynet](https://en.wikipedia.org/wiki/Skynet_(Terminator)) is waging war on humanity". A "**type: bug**" label could not be accurately applied at that same scope.

Rather than having a huge number of specific labels, we use combinations of general labels. So for items that would be classified as "bugs" by a strict definition of the term, we use the combination of the "**type: imperfection**" and "**topic: code**" instead of having a specific single label for that classification. Likewise, and error in the documentation (which is not really a "bug" by a strict definition of the term) would be labeled "**type: imperfection**" and "**topic: documentation**". Project managers also apply "**criticality**" labels to classify the severity

>particularly given the concept of bug is recognised by the Bug Report form

While the wording of the labels was selected for the project maintainers, managers, and developers who are the primary users of the labels, the wording of the issue template chooser was selected for the users and community members who are the primary submitters of issues.

closed 2023-02-15 by per1234

chrisjj, 2023-02-15:
> A "type: bug" label could not be accurately applied at that same scope.

Well obviously. "type: bug" would be accurately applied at the scope of bug.

> So for items that would be classified as "bugs" by a strict definition of the term, we use the combination of the "type: imperfection" and "topic: code" 

That's an oops, I think. Not least because that intersection includes tons of non-bugs.

> While the wording of the labels was selected for the project maintainers, managers, and developers who are the primary users of the labels, the wording of the issue template chooser was selected for the users and community members who are the primary submitters of issues.

I think is worth recognising that a common language is requisite for effective communication.

Thanks.