arduino/arduino-ide #1892: Ability to associate board with sketch

BlaT2512, 2023-02-18:
### Describe the request

It would be great to have the ability for a sketch to be associated with a certain board, so that when the sketch is opened in Arduino IDE it selects the board type for that sketch provided the board is installed and available otherwise it could just fallback to the default behaviour of auto-selecting a board. This could be done with a first-line comment on a sketch containing a board type ID which the IDE would look for. If it sees a sketch with this ID when it is opened it would then select that board type and could auto-select a port if it is recognised with that board type or leave the port selection empty.

### Describe the current behavior

Currently when a sketch is opened in Arduino IDE the IDE automatically selects a port and board, but this is often incorrect especially when multiple different sketches open and multiple boards connected to the computer. By the option for a comment in each sketch on what board it is designed to run on it would assist the auto board selection to be more helpful for users.

### Arduino IDE version

v2.0.3

### Operating system

Windows, macOS

### Operating system version

N/A

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

Nick-Norton, 2023-03-06:
This is a very useful feature to have.
I currently use a plugin called IniIno (https://github.com/tykefcz/IniIno) with the 1.8.x variant of the Arduino IDE.
It stores board config in the Sketch.

dankeboy36, 2024-02-23:
I am interested in implementing this feature. Please add me to the assignees

per1234, 2024-05-20:
Closing as duplicate of https://github.com/arduino/arduino-ide/issues/2438

closed 2024-05-20 by per1234