arduino/arduino-ide #1905: [debug]: {build.openocdscript} recipe is not expanded properly

fpistm, 2023-02-22:
### Describe the problem

Hi,

we currently experiment the debug support within the Arduino IDE 2.x for the [STM32 core](https://github.com/stm32duino/Arduino_Core_STM32).
For this we will provide openocd thanks our package index json file, currently in dev branch:
https://github.com/stm32duino/BoardManagerFiles/blob/dev/package_stmicroelectronics_index.json#L584-L625

Then we have declared the debug part in the platform.txt:
```
# Debugger configuration (general options)
# ----------------------------------------
# EXPERIMENTAL feature:
# - this is alpha and may be subject to change without notice
debug.executable={build.path}/{build.project_name}.elf
debug.toolchain=gcc
debug.toolchain.path={runtime.tools.xpack-arm-none-eabi-gcc-12.2.1-1.2.path}/bin
debug.toolchain.prefix=arm-none-eabi-
debug.server=openocd
debug.server.openocd.path={runtime.tools.xpack-openocd-0.12.0-1.path}/bin/openocd
debug.server.openocd.scripts_dir={runtime.tools.xpack-openocd-0.12.0-1.path}/openocd/scripts
debug.server.openocd.script={build.openocdscript}
```

Finally we have defined the `build.openocdscript` recipe in the boards.txt.
As we have a lot of variant defined, we defined it at variant level:
Example for Nucleo_64 F411RE:

```
################################################################################
# Nucleo 64 boards

Nucleo_64.name=Nucleo-64

Nucleo_64.build.core=arduino
Nucleo_64.build.board=Nucleo_64
Nucleo_64.build.variant_h=variant_{build.board}.h
Nucleo_64.build.st_extra_flags=-D{build.product_line} {build.enable_usb} {build.xSerial}
Nucleo_64.upload.maximum_size=0
Nucleo_64.upload.maximum_data_size=0

# NUCLEO_F411RE board
Nucleo_64.menu.pnum.NUCLEO_F411RE=Nucleo F411RE
Nucleo_64.menu.pnum.NUCLEO_F411RE.node="NODE_F411RE,NUCLEO"
Nucleo_64.menu.pnum.NUCLEO_F411RE.upload.maximum_size=524288
Nucleo_64.menu.pnum.NUCLEO_F411RE.upload.maximum_data_size=131072
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.mcu=cortex-m4
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.fpu=-mfpu=fpv4-sp-d16
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.float-abi=-mfloat-abi=hard
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.board=NUCLEO_F411RE
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.series=STM32F4xx
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.product_line=STM32F411xE
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.variant=STM32F4xx/F411R(C-E)T
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.cmsis_lib_gcc=arm_cortexM4lf_math
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.openocdscript=board/st_nucleo_f4.cfg
```

Unfortunately when we launch the debug, the `build.openocdscript` is not expanded:
```
[2023-02-22T09:50:32.614Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session connected. You can switch to "DEBUG CONSOLE" to see GDB interactions.
"C:\\Users\\<username>\\AppData\\Local\\Arduino15\\packages\\STMicroelectronics\\tools\\xpack-openocd\\0.12.0-1/bin/openocd" -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s "c:\\Users\\<username>\\OneDrive - STMicroelectronics\\Documents\\Arduino\\Blink" -f "C:/IDE/arduino-ide/resources/app/plugins/cortex-debug/extension/support/openocd-helpers.tcl" -f {build.openocdscript}
xPack Open On-Chip Debugger 0.12.0-01004-g9ea7f3d64-dirty (2023-01-30-15:04)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
embedded:startup.tcl:28: Error: Can't find {build.openocdscript}
in procedure 'script' 
at file "embedded:startup.tcl", line 28
[2023-02-22T09:50:32.698Z] SERVER CONSOLE DEBUG: onBackendConnect: gdb-server session closed
GDB server session ended. This terminal will be reused, waiting for next session to start...
```

If we defined it at the root of the section it works:
`Nucleo_64.build.openocdscript=board/st_nucleo_f4.cfg`

I tested with arduino-cli debug and both definition works and `{build.openocdscript}` is porperly expanded. So only the IDE is impacted.

I've also tested to define it at root to use an other recipe (variable) and it does not works with IDE while it works with arduino-cli.
```
Nucleo_64.build.openocdscript={build.openocdscript_ext}
Nucleo_64.menu.pnum.NUCLEO_F411RE.build.openocdscript_ext=board/st_nucleo_f4.cfg
```

One strange behavior if it seems some default recipe are expanded by the IDE but not correctly:
ex: `{build.variant}`
`debug.server.openocd.script={build.variant}/{build.openocdscript}`
is expanded like this:
`-f "STM32C0xx/C031C(4-6)(T-U)/{build.openocdscript}"`

but the selected variant should be `STM32F4xx/F411R(C-E)T`

### To reproduce

define {build.openocdscript} in other way.

### Expected behavior

Same behavior between arduino-cli and the IDE: {build.openocdscript} properly expanded.

### Arduino IDE version

2.0.3

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-02-22:
> I tested with arduino-cli debug and both definition works

Which version of the CLI did you test? IDE2 still using [`0.29.0`](https://github.com/arduino/arduino-ide/blob/4deaf4fb76327c382aea372fb2c24c01b56c2a66/arduino-ide-extension/package.json#L166) CLI.

fpistm, 2023-02-22:
> 

I've tested with the last available release 0.30.0.

fpistm, 2023-02-22:
I've tested with arduino-cli version 0.29.0 and it is properly expanded:
`--file "board/st_nucleo_f4.cfg"`

fpistm, 2024-07-19:
Hi @kittaakos 
Since #2356 is implemented I guess this one is deprecated. I didn't checked if expansion work with new syntax but no more need of it.
So I close this issue.

closed 2024-07-19 by fpistm