arduino/arduino-ide #1911: [debug]: when core is not installed in the default location, debug cannot launched properly

fpistm, 2023-02-22:
### Describe the problem

Hi,
When core is not installed in the default location:
`c:\Users\<username>\AppData\Local\Arduino15`
 debug cannot launched properly with error:
` Error getting Debug info: Unknown FQBN: platform arduino:samd is not installed while it is`

Example with official SAMD core:
[image]

It seems the arduino-cli.yaml available in the default location is ignored by the debug feature. Anyway copying this file belong to the Arduino IDE binary solve the issue.

Note:
- Build and upload works properly.

### To reproduce

Precondition:
Having no SAMD core installed in the default location else it works as it uses the one in the default location instead instead
of the one in the specified directory.

1 - Change the default install location in the `c:\Users\<username>\.arduinoIDE\arduino-cli.yaml` 
ex:
```patch
-  data: c:\Users\<username>\AppData\Local\Arduino15
-  downloads: C:\Users\<username>\AppData\Local\Arduino15\staging
+  data: C:\IDE\portable4ide
+  downloads: C:\IDE\portable4ide\staging
```
2- Launch Arduino IDE
3- Install SAMD core
4- Build a sketch
5- Try to debug, then you got the issue. (See above screenshot)

### Expected behavior

Be able to launch the debug with the core installed in the specified location even if the same core is installed in the default one

### Arduino IDE version

2.0.3

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

fpistm, 2023-02-27:
Maybe related to #122

kittaakos, 2023-02-27:
> Maybe related to #122

Probably not. The VS Code extension must append the `--config-file` flag when generating the debug config. See [here](https://github.com/arduino/vscode-arduino-tools/blob/9c296c4fc83746c29e8b0b9d54fafc01a36b3e33/src/extension.ts#L139).

fpistm, 2023-02-27:
My point was it could be related to the portable mode when we want install to another directory like portable. If it was supported I would probably use it instead of modifying manually the arduino-cli.yaml. 😉

commit dafb2454 (2023-03-29): chore: latest VSIX handles custom directories.data

so when starting the debugger the CLI config path is used by the CLI for
the `daemon -I` command.

Closes #1911