arduino/arduino-ide #1926: Error: The command 'arduino-verify-sketch; cannot be executed. There are no active handlers available for the command.

thorathome, 2023-03-01:
### Describe the problem

Error message came up for the first time on Compile-Update command to Adafruit ESP32 Feather
Newly installed IDE 2.0.4

Never saw this message before.  

### To reproduce

I have two or three windows open, each with different boards and sketches. I have been uploading to COM ports and toNetworked ports. Only saw this message once.  

### Expected behavior

No error message. (or recommendation to user for action). 

### Arduino IDE version

Version: 2.0.4 Date: 2023-02-27T16:14:28.576Z CLI Version: 0.31.0  Copyright © 2023 Arduino SA

### Operating system

Windows

### Operating system version

11 Pro

### Additional context

Multiple windows open. Multiple boards, too. Compile uploads to COM port (ESP32) and the Network Ports (ESP8266). 

Hope this is helpful.  

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

thorathome, 2023-03-01:
[image]

yosob, 2023-10-28:
i meet this problem too, and is there any way to solve this?

Flame344, 2024-07-22:
Is this still left unanswered? I have the exact same error message. If you have found the answer to this, could you please comment it? ty, Lucas.

khawajamechatronics, 2024-08-16:
I am having same issue