arduino/arduino-ide #1935: Light (Theia) hides some text

ninora, 2023-03-05:
### Describe the problem

Editor quick suggestions, part of the highlighted line is invisible as it uses the color as the font.

### To reproduce

Make sure "Editor quick suggestions" is checked in preferences.

### Expected behavior

[image]

### Arduino IDE version

2.0.4

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-03-05:
Hi @ninora. Thanks for your report.

OK, then this is a duplicate of https://github.com/arduino/arduino-ide/issues/1053.

The Arduino IDE project managers have decided that the "Theia" themes should be removed from Arduino IDE (https://github.com/arduino/arduino-ide/issues/1283). So we no longer provide support for these themes.

Please select one of the supported themes from the Arduino IDE preferences

- **Dark (Arduino)**
- **Light (Arduino)**
- **High Contrast (Theia)**

If you discover any problems while using a supported theme, please do let us know.

closed 2023-03-05 by per1234

ninora, 2023-03-05:
Light (Theia) is currently the only option to avoid the **unreadable red on black** of the output window.