arduino/arduino-ide #1936: Provide builds for riscv64 host CPU architecture

N0rbert, 2023-03-05:


anonymouscowhead, 2024-02-15:
tried to build it from source, and it seemed to trip up on nx lacking a native module for riscv64.. https://github.com/nrwl/nx/issues/20914

bitsk, 2024-10-23:
Here is a temporary build version that can run in common RISC-V desktop environments. It only supports AVR development boards, and there are still a large number of packages missing builds. They lack CI, and I'm not sure how to proceed with this in Arduino. [arduino-ide_2.3.3-snapshot-2e78e96b_Linux_riscv64.zip](https://blob.bits.ink/arduino/arduino-ide/arduino-ide_2.3.3-snapshot-2e78e96b_Linux_riscv64.zip)

anonymouscowhead, 2024-11-25:
Excellent, thanks!   Can confirm that the build works on my Lichee Pi 4a with stock OS, and Uno R3 Just Works when compiling, uploading, serial monitor.
Surprisingly the same also works on the R4 board with some apparently-non-fatal errors on the console.  Maybe more progress has been made in the last month on packages needed for the Renesas-based boards?

bitsk, 2025-01-24:
@anonymouscowhead Yes, I recently compiled more toolchains for the RISC-V architecture, which are not yet officially maintained by Arduino. You can find them here: https://blob.bits.ink/arduino/packages/package_index.tar.bz2. Additionally, by adding the following URL to the Board Manager, you can support more architectures boards: https://blob.bits.ink/arduino/packages/package_renesas_1.2.0_index.json
https://blob.bits.ink/arduino/packages/package_rp2040_index.json
https://blob.bits.ink/arduino/packages/package_seeeduino_boards_index.json
https://blob.bits.ink/arduino/packages/package_esp32_index.json
https://blob.bits.ink/arduino/packages/package_sg200x_index.json