arduino/arduino-ide #1937: Can't open COM port when uploading to Arduino Pro Mini

nrc1994, 2023-03-06:
### Describe the problem

After updating to the newest version of the Arduino 2.x.x IDE, I was unable to upload code to my Arduino Pro Mini. When I tried to upload, I got the following error message: 
avrdude: ser_open(): can't open device "\\.\COM9": Access is denied.

Failed uploading: uploading error: exit status 1

This happened no matter what sketch I tried to upload. 

I tried unplugging and plugging back in my FTDI programmer several times, as well as restarting my computer several times and assigning new COM ports to the Arduino. The only time that I was able to get it to upload successfully using IDE version 2.0.4 was when I pressed the reset button at exactly the right time during uploading. I double checked that all my board settings were right (3.3v atmega328 8mhz), and I made sure that I checked "Set RTS on close" in the device manager as specified by the Arduino tutorial for the Pro Mini. 

As a note, I was able to upload sketches to an Arduino UNO board I had laying around with no issues. 

When I re-installed IDE version 1.8.19, I was able to upload my code on the first attempt with no issues. I believe this clearly shows that the issue is a bug in IDE version 2.0.4 

### To reproduce

Attempt to upload a sketch to an Arduino Pro Mini with an Atmega 328 processor, 8mhz clock speed, and 3.3v logic using the FTDI adapter by lowpowerlabs using Arduino IDE version 2.0.4. 

### Expected behavior

The sketch will not upload, and the following error message will be displayed:

 avrdude: ser_open(): can't open device "\\.\COMXX": Access is denied. Failed uploading: uploading error: exit status 1

### Arduino IDE version

2.0.4

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-03-06:
Hi @nrc1994. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1783.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2023-03-06 by per1234

nrc1994, 2023-03-06:
Hello @per1234,

My bad for not seeing that - the workaround described in that report worked as expected.