arduino/arduino-ide #1939: Rename symbol (F2) not working

Minu-IU3IRR, 2023-03-06:
### Describe the problem

Renaming symbol from quick action selection or F2 key opens up the rename function but Shift+enter to preview does not show anything and Enter to appy does not work.

### To reproduce

select any symbol in sketch (variable, function etc...), press F2 or right_ckick->rename, edit name, press enter = nothing

### Expected behavior

select any symbol in sketch (variable, function etc...), press F2 or right_ckick->rename, edit name, press enter = renamed symbol

### Arduino IDE version

2.0.4

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2023-03-07:
Hi @Minu-IU3IRR. Thanks for your report. Which board do you have selected from the Arduino IDE **Tools > Board** menu when you experience this problem?

per1234, 2023-03-07:
I can't reproduce it. What happens if you try to rename the `foo` variable in this sketch?

```cpp
void setup() {
  byte foo;
}
void loop() {}
```

Minu-IU3IRR, 2023-03-07:
the edit prmpt would show and when the new name is written, updon pressing enter, the page would flach and the name does not change. 

UPDATE: i have reinstalled the IDE and manually deleted every single folder on my system between installs for some very wierd behaveor and now everything works, maybe my installation have been corrupted in some way....

closed 2023-03-07 by Minu-IU3IRR